meerk40t/meerk40t #2609: [Bug] 0.9.5b6 Existing ops not reused when you open two files with identical ops

Sophist-UK, 2024-10-01:
If you reclassify then you get duplicated burns.

jpirnay, 2024-10-01:
There has never been a "reuse" ops feature, we loaded all operations just to make sure we had all matching operations. And in all honesty to distinguish whether operations are identical is a non-trivial task as they can differ on a lot of different parameters.
Still I did already merge PR 2602 some days ago (after the release of 0.9.5b7) that would resolve this behaviour.

Sophist-UK, 2024-10-01:
Surely you can create a dunder method to compare two Operations classes for equality of all properties that makes it quite easy.

closed 2024-10-07 by jpirnay

Sophist-UK, 2024-10-07:
So is this actually fixed or closed to get rid of it?