meerk40t/meerk40t #2616: 0.9.5b7: Can drag element in tree and drop on Group node but can't drop on File node

Sophist-UK, 2024-10-03:
Enable File nodes for drop like Group nodes

jpirnay, 2024-10-03:
A file node isn't a group node - it's an indicator from where the child elements came. You can easily change a file node to a regular group node and could do then whatever you like.

Sophist-UK, 2024-10-03:
I know it is a different type of node, but that is not a reason to prevent you dropping elements onto it (when moving them in the tree) in the same way that you can drop elements onto Group nodes.

jpirnay, 2024-10-04:
We agree to disagree, a filenode is a meta node that should not serve as a regular group node.

Sophist-UK, 2024-10-04:
Fair enough. The result is an inconsistent UI though.

closed 2024-10-07 by jpirnay

Sophist-UK, 2024-10-07:
I take it from the discussion that this is not actually completed, but is actually closed as "will not do". As I said above, fair enough.