meerk40t/meerk40t #2617: 0.7.5b7: Can we resequence the context menu of File nodes:?

Sophist-UK, 2024-10-03:
Here is the file node context menu:
[image]

Some of these are common with a Group node context menu:
[image]

To make the UI of these two menus more consistent can we please:
1. Add any entries on the Group menu that still make sense to the File menu?
2. Move any entries unique to the file menu to the top of the menu sequenced in a sensible logical order, with a dividing line separating them from those that are also on the group menu?

Thx.

jpirnay, 2024-10-03:
That makes sense, the context menu entries have grown from a few to quite a lot and could definitely merit from a more logical sequence and optical separation. May I be so bold to ask you to look at the other cases too, so that we can bring some order into them as well. Thank you.

Sophist-UK, 2024-10-03:
> May I be so bold to ask you to look at the other cases too, so that we can bring some order into them as well.

You may certainly be that bold, however sorry but I am not a developer submitting PRs or documentation improvements any more - and until there is sufficient hard evidence that the completely toxic developers clique has been abolished for good and that code and documentation contributions from other members of the community will be welcomed and valued, this will continue to be the case. In the mean time, my contributions are now limited to raising issues and making suggestions, and will continue to be so for the foreseeable future.

closed 2024-10-07 by jpirnay

Sophist-UK, 2024-10-07:
It looks like this one might actually be fixed rather than closed to get rid of it. Apparently now a rare occurence.