meerk40t/meerk40t #2625: [Bug]: Mac OS 13.7 (22H123) - INTEL - not loading

skasilov, 2024-10-07:
### Summary Description

Not loading/starting on Mac OS 13.7 (22H123) - INTEL

### Additional Details

Not loading/starting on Mac OS 13.7 (22H123) - INTEL

### MeerK40t Version

betta 6

### MeerK40t Type

Executable

### Your Operating System

Mac (before M1/Bigsur)

joerlane, 2024-10-07:
Do you see anything when you try to open this version of MeerK40t? Does it simply bounce in the dock and then stop; or does it show anything, or an error of some sort?

You can launch it from terminal by calling the executable directly from inside the .app bundle; that should provide a more specific error message. If MeerK40t is inside the Applications folder, this command will launch it from terminal: /Applications/MeerK40t.app/Contents/MacOS/MeerK40t

skasilov, 2024-10-07:
Good afternoon! 
Thank you so much for your response and willingness to help! 
This is very valuable for us! 
I'm not big expert  for Mac, but I'm a big fan of your program! 
And I really want to continue working with her as I see great potential in her! 
And so what can I say! 
When launching the latest beta version, the program "jumps" in the dock and even starts (a black dot appears that the program is running) 
At the same time, there are no mistakes! 
It's just that the program windows don't appear! 
If you click on the program icon in the dock, you can only close the program "forcibly"
I'll try to run it through the terminal tomorrow and let you know! 
By the way, I also can not launch the previous two beta versions!

Sincerely Sergey
+31617349421
Catharina van Rennesstraat 8,
2511AT, 's-Gravenhage
https://www.linkedin.com/in/sergey-kasilov
https://www.innerquill.com <https://www.innerquill.com/>
https://www.instagram.com/innerquill.project

jpirnay, 2024-10-07:
Just for the avoidance of  doubt: which was the last release that worked for you?

joerlane, 2024-10-08:
I uploaded Beta 6 Mac builds just for your testing. Not likely to make a difference if you say the last version working is Beta 4; but worth a try to narrow things down. 

Curious about launching from the terminal; but not positive it will provide an error if you say MeerK40t is actually opening crashed and not showing any windows. Still worth a try though as I don't have any other quick ideas.

Unfortunately I don't know anyone off hand with an Intel machine running that version of macOS (or even anything close). I moved to apple silicon during macOS 11. 

Ultimately may need to bisect the changes between beta 4 and beta 5 to narrow things down there. There haven't been any changes to the build box during that time; so I'm reasonably sure it's something in those MeerK40t changes.

skasilov, 2024-10-08:
> I uploaded Beta 6 Mac builds just for your testing. Not likely to make a difference if you say the last version working is Beta 4; but worth a try to narrow things down.
>
> Curious about launching [...]

Hi
Thank you very much! 
And where can I download this build for me? 
I don't see it? 
Now I tried to run the latest version again! 
The program did not open, but when I forcibly closed it, I got an error and copied it for you in pdf 
Maybe this will help?
[Uploading error.pdf…]()

skasilov, 2024-10-08:
Ha!
A paradox! 
I just saw that the beta 6 version for mac with intel has appeared! 
I downloaded it and it started! 
And version 7 is still not there!

joerlane, 2024-10-08:
Sounds like you found the Beta 6 that I uploaded for testing earlier; glad to hear that worked.

Not sure what version you did try initially; but Beta 7 is also available here: https://github.com/meerk40t/meerk40t/releases/download/0.9.4070/MeerK40t.dmg

It was tested and proven to be functional in macOS version 13.7 earlier today. Do report back if you continue to have problems. 

I did begin looking through the crash report you sent; but still have a few more things to verify in there; hopefully it won't be necessary to spend significantly more time with it (it's over 700 pages long in that format).

skasilov, 2024-10-09:
Hi! 
Thank you very much for your help and participation! 
And thank you very much for wanting to help - this is very important! 
YES, I've seen this version 7! 
But as I suggested, it does not start and when forced to close it gives an error that I sent! 
But version 6 works great!

Can you check what wrong o different in 7 betta?  

Segey

joerlane, 2024-10-09:
So let me get this straight (since there are some inconsistencies in your reporting so far). Beta 4 works, and Beta 6 works on your computer? But Beta 5 does not, and Beta 7 does not work either?

We tested Beta 7 yesterday on a MacBook Pro with macOS version 13.6.7, and then upgraded to 13.7. It works fine on both macOS versions (included the exact version you specified; 13.7). Not to mention; Beta 5 has been out for almost 3 months now, and this is the first report of any problem. 

Have you tried to run any of the versions that don't otherwise function properly on your system with the Terminal, by specifying the binary executable from INSIDE of the .app bundle as mentioned earlier? I provided a path that would work if it had been installed in your Applications folder. Unfortunately the crash report you sent shows it running from an external disk, or the DMG it's packaged in; not inside the Applications folder, so that path would be different... Running from the terminal may well produce a more useful error message. Again; the path would be INSIDE of the .app bundle like: ./MeerK40t.app/Contents/MacOS/MeerK40t

No one else is able to reproduce the problem you are reporting; even on the exact same version of macOS on Intel hardware.

jpirnay, 2024-10-09:
"Have you tried to turn it on and off again?" (The IT crowd)
Have you tried to nuke your settings (aka move them somewhere else for the test) ?

skasilov, 2024-10-09:
Hi! 
Thank you so much for wanting to help! 
I'm sorry - I think I didn't give you tight information! 
I've run all versions up to beta 6! 
Beta 5 open but I had problems with engraving and I returned to version 4!
Then version 6 did not run for me, but your last change will allow it to run and I have been working with it for a couple of days without gaps! 
Version 7 still doesn't launch! 
I tried to place the file on different disks - nothing! 
Icon jumps in the dock - a dot appears under icon (like for running app) dock and that's it! 
The windows won't open! 
You make a forced exit - an error message appears that I sent! 
Tomorrow I will try to launch the application through the terminal = and let you know! 
Crazy idea - I have several monitors connected - no connection to this? 
Very strange! 
Version 6 didn't run before - you did something - now everything works :-))))
Let's do this with version 7? :-))))

Sergey

skasilov, 2024-10-10:
Hi! 
This is log from terminal start 7 beta 
Hioe it will help! 
[Uploading TERMINAL.pdf…]()

skasilov, 2024-10-10:
[TERMINAL.pdf](https://github.com/user-attachments/files/17327767/TERMINAL.pdf)

joerlane, 2024-10-10:
Thank you very much. That's a much more coherent error message. I could tell from the earlier crash report that it was throwing an error; but that error wasn't reported directly in the crash report. Running it in terminal as you just did, printed out that python/system error. 

Looking into it now. Much appreciated as we were unable to reproduce that error, even with the same version of macOS you reported using.

joerlane, 2024-10-10:
That was a recent commit. The error will be fixed. 

I do want to ask a few questions for my own edifications (so that I can better help in the future). What hardware are you running on? I saw that Lilu was installed; and that stalled me for a bit. I know what it does. But it gave me pause. I'm well aware of what it can do; and what it is used for. 

Also considered the multiple display issue. But that doesn't cause a crash; simply might lose windows in certain rare circumstances. And those can be fixed by dumping the settings, or editing them with a text editor. 

Finally; what kind of laser hardware are you using? We support a lot of machines. That question is mostly out of curiosity; not at all related to the bud report. 

Should have a solutions soon.

Again. Very much appreciate you using the terminal to check here. That is very much more helpful with certain Mac issues.

joerlane, 2024-10-10:
Beta 8 has been release in your honor. Please let me know if you have any further issues.

skasilov, 2024-10-11:
Good afternoon! 
Thank you so much for your efforts and willingness to help! 
You're really doing a great job and I'm a fan of yours.  :-) 
The program has excellent prospects :-)
You'll see and one day overtake LightBurn :-)
I will check version 8 and let you know today! 
But as I understand it, did you find the reason why version 7 did not launch? 

According to your questions!
Lilu - well, it's a firewall :-)
It's just that sometimes I don't want some programs to go where they don't need to without your knowledge. :-) So we use it! 

We use XTOOL lasers and their programs in our work mainly! 
But we also have K 40 and for him your program is simply irreplaceable! 
We also have a diode laser https://www.toocaa.com
We want to try it with him! 
By the way, are you polishing to add camera work? 
Will there be such a feature?
It would be very interesting!
Thank you for your work! 

Sergey

skasilov, 2024-10-11:
Just for info! 
8 beta work perfect! 
Thank you!!!

jpirnay, 2024-10-11:
Good to hear

closed 2024-10-11 by jpirnay