meerk40t/meerk40t #2626: [Bug]: Laser not recognized if plugged in after program start (Linux)

jpirnay, 2024-10-07:
### Summary Description

@GazButane
Author
[GazButane](https://github.com/GazButane) commented [on Aug 22](https://github.com/meerk40t/meerk40t/issues/new?assignees=&labels=Type%3A+Bug&projects=&template=bug_report.yml&title=%5BBug%5D%3A+#issuecomment-2304991607)
Hello, thank you for your answer
I did some tests with the latest version and the bug I encountered previously seems not to have recurred! Following these tests I still noticed that when I plug the laser into the computer AFTER starting Meerkat, the USB port is not detected automatically and it is necessary to select it manually in the settings. no problem however when I plug the laser BEFORE opening Meerkat

### MeerK40t Version

0.95x

### MeerK40t Type

Executable

### Your Operating System

Linux

Sophist-UK, 2024-12-14:
My 3d printer is driven by Octopi under Ubuntu Linux on a rPi400, and I had a very similar issue to this. 

I cannot remember the details of how, but based on research on t'internet I found a way to configure Linux to inform Octopi when the printer was plugged in.

Sophist-UK, 2025-06-24:
I didn't have any issue with the bot closing this - if Octoprint cannot automatically detect a USB connection and needs the user to configure it, I don't see why Meerk40t should be any different.

So if anything is needed, it is probably documentation.