meerk40t/meerk40t #2634: [Bug]: Moshiboard does not reflect speed settings correctly

mnoriaki, 2024-10-11:
### Summary Description

Moshiboard moves and cuts using previous speed setting. For example, if you press home button in navigation panel before starting an operation, it moves in 40mm/s regardless of the speed settings. The speed at the end of the first operation will be used when continuing with the second operation.

### Additional Details

Attached is the log file [20241011.txt](https://github.com/user-attachments/files/17337236/20241011.txt) opening Moshibord/events channel when I try to cut and engrave two lines as the captured window below. Mock USB driver is used. The blue line should be cut in 10mm/s and the red one should be cut in 1 mm/s. I pressed Arm-Start, Arm-Start, Home (in navigation panel). To summarize the log by extracting only the speed line; 

> Arm-Start
> [12:21:07]     Vector Cut Speed: 10 mm/s Normal Speed: 10 mm/s
> [12:21:07]     Vector Cut Speed: 10 (09) mm/s Normal Speed: 10 mm/s (09)
> Arm-Start
> [12:21:14]     Vector Cut Speed: 1 mm/s Normal Speed: 1 mm/s
> [12:21:14]     Vector Cut Speed: 1 (00) mm/s Normal Speed: 1 mm/s (00)
> Home
> [12:21:23]     Vector Cut Speed: 40 mm/s Normal Speed: 40 mm/s
> Arm-Start
> [12:21:27]     Vector Cut Speed: 40 mm/s Normal Speed: 40 mm/s
> [12:21:27]     Vector Cut Speed: 40 (27) mm/s Normal Speed: 40 mm/s (27)
> Arm-Start
> [12:21:31]     Vector Cut Speed: 1 mm/s Normal Speed: 1 mm/s
> [12:21:31]     Vector Cut Speed: 1 (00) mm/s Normal Speed: 1 mm/s (00)

[image]

I guess this bug is caused by the fact that it is expected to reflect the speed and other settings by the line 309 meerk40t/moshi/driver.py :
`            self.settings.update(q.settings)`
but actually it is not. Although the self.settings.speed is updated, either vector_speed() or raster_speed() in meerk40t/moshi/builder.py is not called.

Here is quick hack to fix this problem: [20241011-patch.txt](https://github.com/user-attachments/files/17337799/20241011-patch.txt). However, I don't think it is good enough to be merged. I disabled some assertion and it is not a good idea to set speed for every movement. 

### MeerK40t Version

v0.9.4050

### MeerK40t Type

Executable

### Your Operating System

Mac (M1/Bigsur)

jpirnay, 2024-10-13:
Thx for spotting this, don't know why and when a very relevant piece of code disappeared in the queue handling... But anyway PR #2641 should fix this

jpirnay, 2024-10-13:
Could you please check this on your real moshi - as I would need to update a couple of unittests

mnoriaki, 2024-10-15:
I have tested with our Moshiboard K40 and it worked fine.

Thank you!

closed 2024-10-15 by jpirnay