meerk40t/meerk40t #2639: 0.9.5b8: Feedback on new tree context menus after #2619

Sophist-UK, 2024-10-12:
@jpirnay Jens

#2619 was an excellent improvement and the code looks very elegant too.

A few comments on the menus as they now look:
### General
* Most menu items are Sentence ***c***ase - but a few are Title ***C***ase
* Many status bar hints are missing when you hover over a context menu
### Operations top-level node
[image]
* Global ***S***ettings - Not clear what these are from the name - Global operations properties?
* Classification –duplicate entry for Refresh classification for all - 
   [image] 
* Scale Speed/Power Settings - Name and submenu contents are conflicting - menu says "Scaling" suggesting that this scales the speeds or powers set on individual operations but sub-menu talks about Maximums.

### Image operation node
[image]
* (For all operations nodes) Duplicate element(s)/operations - Perhaps in the same menu group?
* (For all operations nodes) Duplicate element(s) - not actually duplicating the elements themselves - more like Duplicating operation's burns on those elements

### Raster operation node
[image]
No additional comments

### Engrave operation node
[image]
No additional comments

### Cut operation node
No additional comments

### Elements top-level node

### File elements node
1. Toggle Visibility on File node doesn't do anything.

### Group elements node

### Image elements node

### Path elements node

### Shape elements node

NOTE: Full description to be completed.

jpirnay, 2024-10-13:
Addressed in PR #2640 

1.     Case of menu items aligned unless the term has a special meaning like Cut or Raster
2.      Help text content added for all items
3.      Some terms changed to provide more clarity for certain menu items 8GLobal ops properties, Scale speed / power)
4.      The menu items for an element under an operation consist of the reference specific and underneath those belonging to the referred element. These two can't be easily mixed without addtional refactoring.

Sophist-UK, 2024-10-13:
4. Not sure which item I reported that this relates to, but all of my comments were language related and not functionally related.

closed 2024-10-29 by jpirnay