meerk40t/meerk40t #2643: [Suggestion] 0.9.5b8: Splashscreen

Sophist-UK, 2024-10-15:
I absolutely love the splash screen but...

1. It would be even better with the word "Meerk40t" incuded.
2. It would be even better with the version overlayed in the bottom RH corner.

jpirnay, 2024-10-17:
The text part of the screen does display Meerk40t plus version information in the beginning - I could display it all the time throughout the startup phase

Sophist-UK, 2024-10-17:
I don't see any words or versions when loading on Windows. Sorry.

jpirnay, 2024-10-17:
Hmm, don't you see this:
[image]

Sophist-UK, 2024-10-17:
No - the original pic I posted is all I see.

jpirnay, 2024-11-04:
Being displayed and confirmed as working.

closed 2024-11-04 by jpirnay