meerk40t/meerk40t #2654: [Bug]: Title texts aren't very visible

Laserology, 2024-10-22:
### Summary Description

Using black and white gradient for the background for title bars isn't a good idea, as both white and dark text will be not as visible. What other color combos could be tried that would be ok?
My idea would be white+grey on light mode, and black+grey on dark mode.
[image]

### Additional Details

Also, this is just me personally, but gradients feel rather outdated in my opinion, at least when used in this context.

### Crash logs

```shell
N/A
```

### MeerK40t Version

N/A

### MeerK40t Type

PIP3/Pypi Package

### Your Operating System

Linux

jpirnay, 2024-10-22:
Enhancement request - this is deep inside the ArtProvider for wx.aui. As for the api for title bars this has been included - so I would like to see a screenshot please as of the latest source from github. The ArtProvider for notebook tabs is very flawed - it neither exposes the gradient property to be changed nor does it work consistently (you can change the colors if the tab indicators are on the bottom, you can't if they are on the top) - so this needs to go to the backlog for now

Laserology, 2024-10-22:
> Enhancement request - this is deep inside the ArtProvider for wx.aui. As for the api for title bars this has been included - so I would like to see a screenshot please as of the latest source from [...]

Im not sure exactly what you need a screenshot of - i am not very familiar with the terminology here. But the screenshot i provided was indeed from latest git, by mistake i labeled it as being the pip package.

jpirnay, 2024-10-22:
>  But the screenshot i provided was indeed from latest git

I got that - I wanted to see whether the issue is still in place after the last comitts. Just downloading Fedora Kinoite to set up a VM for myself

jpirnay, 2024-10-23:
Gradients are no longer in place for window titles, admittedly they are for notebook tabs, but that is outside our control.
[image]

closed 2024-10-23 by jpirnay

Laserology, 2024-10-23:
I think the tabs look okay anyways.
Though i notice in the screenshot, they use the same color as the background, would it be better to use a different color for contrast?