meerk40t/meerk40t #2655: [Bug]: Manual laser move buttons stay highlighted after being hovered once

Laserology, 2024-10-22:
### Summary Description

In the 'Jog' menu, when hovering over buttons, their hover state does not update once un-hovered.

### Additional Details

No other buttons have this issue so it is likely an easy fix.

### Crash logs

```shell
N/A
```

### MeerK40t Version

v0.9.5 git - latest commit

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Linux

jpirnay, 2024-10-22:
I don't see this - please paste the System information / Component information from the "About" dialog to help me better understand your environment

Laserology, 2024-10-22:
> I don't see this - please paste the System information / Component information from the "About" dialog to help me better understand your environment

MeerK40t v0.9.5000 git main
/home/[...]/.config/MeerK40t
System: Linux
Node Name: toolbx
Release: 6.11.3-200.fc40.x86_64
Version: #1 SMP PREEMPT_DYNAMIC Thu Oct 10 22:31:19 UTC 2024
Machine: x86_64
Processor: 
Theme: system, Darkmode: True
Ip-Address: localhost

This is mostly generic due to being run in a development container - but my core system details still apply anyways.
I'll add them here myself:
KDE plasma 6.2.1
QT 6.7.2
Fedora Kinoite 41
Plasma Style: Utterly round solid

If you need more info, let me know.

jpirnay, 2024-10-22:
Okay, and the wxPython version?

Laserology, 2024-10-22:
> Okay, and the wxPython version?

4.2.2 (gtk3, Phoenix) wxWidgets 3.2.6

jpirnay, 2024-10-23:
I can't get Fedora Kinoite to work but I checked on a couple of different Linux distros and I am not seeing the reported issue on any of them. So a bit helpless, I have to admit.

Laserology, 2024-10-23:
> I can't get Fedora Kinoite to work but I checked on a couple of different Linux distros and I am not seeing the reported issue on any of them. So a bit helpless, I have to admit.

And to confirm, you tested the jog/move buttons, correct?

jpirnay, 2024-10-23:
[image]

Laserology, 2024-10-23:
> ![jogmove](https://private-user-images.githubusercontent.com/2670784/379428690-b20253ec-eacb-4c93-bc03-02124d4a1eae.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MTA2OTYsIm5iZiI6MTcyOTcxMDM5NiwicGF0aCI6Ii8yNjcwNzg0LzM3OTQyODY5MC1iMjAyNTNlYy1lYWNiLTRjOTMtYmMwMy0wMjEyNGQ0YTFlYWUuZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMTkwNjM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWVhNzQzZDUyZjBmNDEzMTRhMmQxZDYyZmE0ZDlkMmJmZTdkMDkxZGU0YmFkYzBmMDNlZWIyY2VkMWM3MGJlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ZfsyrO4rcy-WdOR1CgNPeRyipFrxKoWOoacawjSyCGo) [ ![jogmove](https://private-user-images.githubusercontent.com/2670784/379428690-b20253ec-eacb-4c93-bc03-02124d4a1eae.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MTA2OTYsIm5iZiI6MTcyOTcxMDM5NiwicGF0aCI6Ii8yNjcwNzg0LzM3OTQyODY5MC1iMjAyNTNlYy1lYWNiLTRjOTMtYmMwMy0wMjEyNGQ0YTFlYWUuZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMTkwNjM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWVhNzQzZDUyZjBmNDEzMTRhMmQxZDYyZmE0ZDlkMmJmZTdkMDkxZGU0YmFkYzBmMDNlZWIyY2VkMWM3MGJlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ZfsyrO4rcy-WdOR1CgNPeRyipFrxKoWOoacawjSyCGo) ](https://private-user-images.githubusercontent.com/2670784/379428690-b20253ec-eacb-4c93-bc03-02124d4a1eae.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MTA2OTYsIm5iZiI6MTcyOTcxMDM5NiwicGF0aCI6Ii8yNjcwNzg0LzM3OTQyODY5MC1iMjAyNTNlYy1lYWNiLTRjOTMtYmMwMy0wMjEyNGQ0YTFlYWUuZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMTkwNjM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWVhNzQzZDUyZjBmNDEzMTRhMmQxZDYyZmE0ZDlkMmJmZTdkMDkxZGU0YmFkYzBmMDNlZWIyY2VkMWM3MGJlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ZfsyrO4rcy-WdOR1CgNPeRyipFrxKoWOoacawjSyCGo) [ ](https://private-user-images.githubusercontent.com/2670784/379428690-b20253ec-eacb-4c93-bc03-02124d4a1eae.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MTA2OTYsIm5iZiI6MTcyOTcxMDM5NiwicGF0aCI6Ii8yNjcwNzg0LzM3OTQyODY5MC1iMjAyNTNlYy1lYWNiLTRjOTMtYmMwMy0wMjEyNGQ0YTFlYWUuZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMTkwNjM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWVhNzQzZDUyZjBmNDEzMTRhMmQxZDYyZmE0ZDlkMmJmZTdkMDkxZGU0YmFkYzBmMDNlZWIyY2VkMWM3MGJlMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ZfsyrO4rcy-WdOR1CgNPeRyipFrxKoWOoacawjSyCGo)

I can see them being highlighted, the system you tested on just doesn't make it very obvious. You can see the blue border around the buttons appear after you hover them, and they don't go away afterwards.

jpirnay, 2024-10-23:
Yeah, a different theme helped! Fixed
[image]

closed 2024-10-23 by jpirnay

Laserology, 2024-10-23:
> Yeah, a different theme helped! Fixed 

I just noticed in my own instance... what happened to the color themes? The bed is an off-purple now, the grid lines are invisible, and the guide markers are an odd cyan color. I don't see it changed anywhere in the commits, the colors chosen don't seem that great in terms of contrast and simplicity

Laserology, 2024-10-23:
The color preferences don't seem so save either

PR #2663 (2024-10-23): Fix hover issue on jog panel
Propagate previously swallowed events, fixes #2655

## Summary by Sourcery

Bug Fixes:
- Fix hover issue on jog panel by propagating previously swallowed events.