meerk40t/meerk40t #2657: [Bug]: Random file created in home directory

Laserology, 2024-10-22:
### Summary Description

When launched, the program creates a file in the home directory with a size of 6 bytes. This should be removed, as it does not appear to have a use.

### MeerK40t Version

N/A

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Linux

jpirnay, 2024-10-22:
Again, I don't see this - which filename, which content?

Laserology, 2024-10-22:
> Again, I don't see this - which filename, which content?

Pardon my "bombing" of issues - wanted to get things noted down before night so i didn't have much time to complete the issues.
Filename is MeerK40-{username} (in /home/{username}/)
Contents are unknown, looks maybe like a process ID, it's some numbers.

Laserology, 2024-10-22:
> This is PID file for the SingleInstanceChecker - that's valid and should not exist beyond the program scope.

Why not check the process list instead?

And why not move the file? It should probably be under config, not flooding the user's home directory.

Also, the file seems to persist even when the app is closed.

Overal the implementation is not great

jpirnay, 2024-10-22:
Shrug, I don't want to dismiss your comment, but please remember MeerK40t is multi-platform and we tend to use the functionalities provided by our most used framework aka wxPython before reinventing the wheel / adding additional libraries.

Laserology, 2024-10-23:
> Shrug, I don't want to dismiss your comment, but please remember MeerK40t is multi-platform and we tend to use the functionalities provided by our most used framework aka wxPython before reinventing the wheel / adding additional libraries.

Would python not have the ability to do a feature like this? I don't imagine it being too hard. Just write to an empty folder called ``running.lock`` in the config folder, and delete it when the program closes.

jpirnay, 2025-02-25:
May have found a way to redirect it. Part of next release

Sophist-UK, 2025-02-25:
IME it is normal to put lock files like this in `/tmp` on Linux systems so that it doesn't survive a reboot.