meerk40t/meerk40t #2660: [enhance]: Misc repo enhancments

Laserology, 2024-10-23:
Just some ideas to broaden the scope of the project and attract more users/contributors:
- Update the screenshots section once theming updates are complete
- Change the repo's description to be more general, not just for k40 lasers
- Finalize 9.5 release when GUI changes are finished (?)
- Update project tags (categories)
- Hide the 'packages' tab on the front page - not used, causes unwanted clutter

These are minor details, but play a significant role in how the project is seen, as they are essentially front and center when users visit the GitHub page. Changing these would likely be beneficial for the project.

Thoughts on these changes?

jpirnay, 2024-10-23:
Made it in a list and adjusted somethings already

Laserology, 2024-10-23:
> Made it in a list and adjusted somethings already

Great, looks good so far!

Laserology, 2025-05-08:
This is still relevant