meerk40t/meerk40t #2665: [Bug]: Color themes broken in recent commits

Laserology, 2024-10-23:
### Summary Description

I don't know since when, but the latest commits have broken color themes. They no longer use the colors specified in the default profile, and they do not save when a user changes them and reopens the program.

### Additional Details

Odly-pickeed colors are now used for Bed, grid, etc...

### Crash logs

```shell
N/A
```

### MeerK40t Version

v0.9.5 git - latest commit

### MeerK40t Type

Git

### Your Operating System

Linux

Laserology, 2024-10-23:
Top bar buttons also no longer follow the requested icon size
[image]

jpirnay, 2024-10-23:
> Top bar buttons also no longer follow the requested icon size

The buttons in the ribbon bars (top + left) scale automatically. The setting for icons is about the icons in the regular buttons

jpirnay, 2024-10-23:
> Odly-pickeed colors are now used for Bed, grid, etc...

Those were from your PR - I don't like them either

jpirnay, 2024-10-23:
The saving / loading bug is acknowledged and fixed

jpirnay, 2024-10-23:
The ugly defaults for black mode remain

Laserology, 2024-10-23:
> > Odly-pickeed colors are now used for Bed, grid, etc...
> 
> Those were from your PR - I don't like them either

I don't recall ever changing the bed to a purple color... I changed it to use different menu colors that would blend into the system, not stick out like a sore thumb

Laserology, 2024-10-23:
> > > Odly-pickeed colors are now used for Bed, grid, etc...
> > 
> > Those were from your PR - I don't like them either
> 
> I don't recall ever changing the bed to a purple color... I changed it to use different menu colors that would blend into the system, not stick out like a sore thumb

Can you point to a specific change where I set this?

jpirnay, 2024-10-23:
[image]

Laserology, 2024-10-23:
> [image]
> 
> 

When I tested that, it worked just fine. Are you sure nothing else changed?

Laserology, 2024-10-23:
I see now somebody added 'tohex()'... Is that needed? I didn't have any issues when I tested originally

Sophist-UK, 2024-10-23:
I cannot believe that it is me saying this, but let's try to remember that we are all of us human and can make mistakes or write imperfect code - and that we all have a common goal, to make MK better and better. So it might be better to worry more about how to fix something that has gone wrong, and less about who caused it to go wrong.

jpirnay, 2024-10-23:
It's partly you and partly me: your code crashed on non linux and i added the hex - but its wrongly coded. Just fixing it

closed 2024-10-23 by jpirnay

Laserology, 2024-10-23:
> I cannot believe that it is me saying this, but let's try to remember that we are all of us human and can make mistakes or write imperfect code - and that we all have a common goal, to make MK [...]

It is not my intention to spark debate over who did it - apologies if it seemed that way.

I have autism, which (for me) means I have a difficult time communicating, vocabularies are sometimes difficult and I often use words or entire sentence structures interchangeably. I did not intend to pin the issue on any individual, just a mistake of wording on my part.

Again, apologies to everyone, and thank you for pointing out my mistake.

jpirnay, 2024-10-23:
I can assure you, no harm was detected or felt. We are happy to have you around

Sophist-UK, 2024-10-24:
> It is not my intention to spark debate over who did it - apologies if it seemed that way.
>
> I have autism, which (for me) means I have a difficult time communicating, vocabularies are sometimes [...]

I assure everyone here that I was NOT pointing the finger of blame at any individual for blaming others themselves - I am not *that* hypocritical.