meerk40t/meerk40t #2673: [Bug]: Core dump on adding text element to workspace

Laserology, 2024-10-24:
### Summary Description

When attempting to add text to a project, the program crashes, simply stating "Segmentation fault (core dumped)" in the console.

### Additional Details

Open new project
Add text
See crash

### Crash logs

```shell
Unsure where to get them
```

### MeerK40t Version

v0.9.5 git - latest commit

### MeerK40t Type

Git

### Your Operating System

Linux

Laserology, 2024-10-24:
Update: Program now spat out this info:
```
Gtk:ERROR:../gtk/gtktextbtree.c:1545:_gtk_text_btree_find_line_top: code should not be reached
Bail out! Gtk:ERROR:../gtk/gtktextbtree.c:1545:_gtk_text_btree_find_line_top: code should not be reached
```

jpirnay, 2024-10-24:
No idea why this happens, but I can avoid it. My take: wxpython under Linux crashes hard when a control is assessed while the containing frame is still being intiated on the display system. The culprit code is simply: 'Focus a control, and select all text in it'.
The workaround is to wait a bit and then do exactly the same thing. It's things like these that drive me crazy....

Laserology, 2024-10-24:
> No idea why this happens, but I can avoid it. My take: wxpython under Linux crashes hard when a control is assessed while the containing frame is still being intiated on the display system. The [...]

Would there be some check you can do to see if the UI is ready?

jpirnay, 2024-10-24:
I did try that, wx.Window the baseclass exposes a couple of indicators:
IsShown, IsShownOnScreen, IsFocusable
all three were positive and it still crashed...

jpirnay, 2024-10-24:
<rant>wxpython has more bugs than MeerK40t - and that is something you shouldn't be proud of</rant> ;-)

Laserology, 2024-10-24:
How long am I supposed to wait for the window to be done loading? I waited about 45 seconds and it still crashed

jpirnay, 2024-10-24:
Really? The try to return directly without the tect selection

Laserology, 2024-10-24:
> Really? Then try to return directly without the text selection 

 Return to what?

jpirnay, 2024-10-25:
I have fully eliminated the code for Linux in PR #2676

closed 2024-10-25 by jpirnay