meerk40t/meerk40t #2687: [Bug]: since 9.5 meerk40t fail to connect to my M2nano controler

cminssie, 2024-10-28:
### Summary Description

9.4 is connecting natively in usb without problems.
Is there something that should be changed in config grom 9.4 to 9.5 ? (Windows 64bits)

### MeerK40t Version

9.5

### MeerK40t Type

Executable

### Your Operating System

Windows

cminssie, 2024-10-29:
> The 64 bit version might still face issues - please fall back to the regular (32bit) version (meerk40t.exe) - but: please share the output being displayed in the controller window.
> And please try meerk40t-win-all.exe too

Hi,Yes the 32 bits release connect well.Thanks for the advise.This product is great value.
Christian Minssieux06 89 88 25 ***@***.***
316 chemin de Chaillonnette38460 Crémieu

jpirnay, 2024-11-04:
Due to experimental win64 compilation - we have one instance where we can create a working win64 executable - so we still need to figure out what's different on that local machine vs the github virtual instance

jpirnay, 2024-11-04:
Wont build win64 versions for now.

closed 2024-11-04 by jpirnay