meerk40t/meerk40t #2693: [Bug]: MAC OS 13.7.1 (22H221) INTEL - 0.9.5000 - CRASH

skasilov, 2024-10-30:
### Summary Description

Hi! 
[Bug]: MAC OS 13.7.1 (22H221) INTEL - 0.9.5000 - CRASH
This LOG 

### Crash logs

```shell
MeerK40t crash log. Version: 0.9.5000 on Darwin: Python 3.8.7: x86_64 - wxPython: 4.2.1 osx-cocoa (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "mk40t.py", line 11, in <module>
  File "meerk40t/main.py", line 198, in run
  File "meerk40t/main.py", line 230, in _exe
  File "meerk40t/kernel/kernel.py", line 1150, in __call__
  File "meerk40t/kernel/kernel.py", line 860, in set_kernel_lifecycle
  File "meerk40t/gui/plugin.py", line 411, in plugin
  File "meerk40t/kernel/kernel.py", line 2296, in console
  File "meerk40t/kernel/kernel.py", line 2342, in _console_parse
  File "meerk40t/kernel/functions.py", line 286, in inner
  File "meerk40t/gui/wxmeerk40t.py", line 708, in window_open
  File "meerk40t/gui/wxmeerk40t.py", line 697, in window_open
  File "meerk40t/kernel/context.py", line 419, in open_as
  File "meerk40t/gui/wxmmain.py", line 243, in __init__
  File "meerk40t/gui/mwindow.py", line 36, in __init__
  File "meerk40t/gui/themes.py", line 150, in set_window_colors
wx._core.wxAssertionError: C++ assertion ""IsOk()"" failed at /private/var/folders/ng/qbrmmqsx6990j5cfzz6t_z0h0000gn/T/pip-install-x53_dp_9/wxpython_8de19be4eb1141a98a5a5d6b51ee1eb4/ext/wxWidgets/src/osx/core/colour.cpp(219) in OSXGetNSColor(): invalid colour

Local variables:
[mk40t.py:11, in <module>]:
__name__ : __main__
__doc__ : None
__package__ : None
__loader__ : <class '_frozen_importlib.BuiltinImporter'>
__spec__ : None
__annotations__ : {}
__builtins__ : <module 'builtins' (built-in)>
__file__ : /Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/mk40t.py
sys : <module 'sys' (built-in)>
pyimod03_importers : <module 'pyimod03_importers' from 'PyInstaller/loader/pyimod03_importers.py'>
os : <module 'os' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/os.pyc'>
VIRTENV : VIRTUAL_ENV
python_path : ['/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/base_library.zip', '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/lib-dynload', '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS']
pth : /Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS
NullWriter : <class '__main__.NullWriter'>
encodings : <module 'encodings' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/base_library.zip/encodings/__init__.pyc'>
ctypes : <module 'ctypes' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/ctypes/__init__.pyc'>
LibraryLoader : <class 'ctypes.LibraryLoader'>
DEFAULT_MODE : 4
_frozen_name : <function _frozen_name at 0x7fd56850ab80>
PyInstallerImportError : <class '__main__.PyInstallerImportError'>
PyInstallerCDLL : <class '__main__.PyInstallerCDLL'>
PyInstallerPyDLL : <class '__main__.PyInstallerPyDLL'>
dyld : <module 'ctypes.macholib.dyld' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/ctypes/macholib/dyld.pyc'>
d : /Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/eggs
re : <module 're' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/base_library.zip/re.pyc'>
multiprocessing : <module 'multiprocessing' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/multiprocessing/__init__.pyc'>
spawn : <module 'multiprocessing.spawn' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/multiprocessing/spawn.pyc'>
_args_from_interpreter_flags : <function _args_from_interpreter_flags at 0x7fd56905d790>
_freeze_support : <function _freeze_support at 0x7fd568571ee0>
forking : <module 'multiprocessing.popen_fork' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/multiprocessing/popen_fork.pyc'>
spawning : <module 'multiprocessing.popen_spawn_posix' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/multiprocessing/popen_spawn_posix.pyc'>
FrozenSupportMixIn : <class '__main__.FrozenSupportMixIn'>
_Popen : <class '__main__._Popen'>
_Spawning_Popen : <class '__main__._Spawning_Popen'>
tcldir : /Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/tcl
tkdir : /Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/tk
is_darwin : True
pathlib : <module 'pathlib' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/pathlib.pyc'>
pkg_resources : <module 'pkg_resources' from '/Users/kasilov/Desktop/MeerK40t.app/Contents/MacOS/pkg_resources/__init__.pyc'>
[... 162 lines cut ...]
width : 1728
height : 972
context : Context('/')
path : window/MeerK40t
parent : None
style : 541072960
args : ()
kwds : {}
[meerk40t/gui/themes.py:150, in set_window_colors]:
self : Service('themes', None)
win : MeerK40t(Context('/'), name="window/MeerK40t")
tp : {'win_bg': None, 'win_fg': None, 'button_bg': None, 'button_fg': None, 'text_bg': None, 'text_fg': None, 'list_bg': None, 'list_fg': None, 'label_bg': None, 'label_fg': None, 'highlight': None, 'inactive_bg': None, 'inactive_fg': None, 'pause_bg': None, 'pause_fg': wx.Colour(0, 0, 0, 216), 'start_bg': None, 'start_fg': None, 'start_bg_inactive': None, 'start_fg_focus': None, 'stop_bg_inactive': None, 'stop_bg': None, 'stop_fg': None, 'stop_fg_focus': None, 'arm_bg': None, 'arm_fg': None, 'arm_bg_inactive': None, 'arm_fg_focus': None}
```

### MeerK40t Version

0.9.5000

### MeerK40t Type

Git

### Your Operating System

Mac (before M1/Bigsur)

jpirnay, 2024-10-30:
This is on startup? Can you please provide a bit more context - thank you

skasilov, 2024-10-30:
Good afternoon! 
Thank you so much for your response and willingness to help! 
I'm sorry - I just thought it was clear from this log. 
When the application starts, this error is displayed with a suggestion to send a report to the developers! 
When choosing ANY answer YES or NO, the application freezes dead! 
Only forcibly closing the application helps! 
Previous versions work fine! 
I really hope that you will solve the problem!

Sincerely Sergey Kasilov
CEO/CTO
+31617349421

The Netherlands
Catharina van Rennesstraat 8,
2511AT, 's-Gravenhage
https://www.linkedin.com/in/sergey-kasilov
https://www.innerquill.com
https://www.etsy.com/shop/InnerQuill
https://www.instagram.com/innerquill.project

Unique tools for creativity and calligraphy.
3D printing - 3D scanning - Laser Engraving Personalization.

￼

jpirnay, 2024-10-30:
I assumed so, but I am always happy to receive as much context as possible.
Are you able to run from source? Rhen I would like to see whether the issue is still in place with the latest github code.

skasilov, 2024-10-30:
Thanks for the answer! 
No! 
As usual, I downloaded the INTEL Mac version - I moved it to the "programs" folder
and even tried to transfer it to another disk! 
The result is the same - an error when starting and freezing the program!

Sincerely Sergey Kasilov
CEO/CTO
+31617349421

The Netherlands
Catharina van Rennesstraat 8,
2511AT, 's-Gravenhage
https://www.linkedin.com/in/sergey-kasilov
https://www.innerquill.com
https://www.etsy.com/shop/InnerQuill
https://www.instagram.com/innerquill.project

Unique tools for creativity and calligraphy.
3D printing - 3D scanning - Laser Engraving Personalization.

￼

jpirnay, 2024-10-31:
I would be grateful for your meerk40t.cfg file, to be able to check what's going on.
(You could try to move that file to a different location to force mk to start from scratch, not using any stored settings, does the crash then still happen?)
Are you running on proper Apple Hardware or on a hackintosh style system?

skasilov, 2024-10-31:
Hi! 
Thank you so much for your help! 
Can you help me - where to look for this meerk40t.cfg file? 
The search doesn't show anything!! 
I have an original Mac mini

Sincerely Sergey Kasilov
CEO/CTO
+31617349421

The Netherlands
Catharina van Rennesstraat 8,
2511AT, 's-Gravenhage
https://www.linkedin.com/in/sergey-kasilov
https://www.innerquill.com
https://www.etsy.com/shop/InnerQuill
https://www.instagram.com/innerquill.project

Unique tools for creativity and calligraphy.
3D printing - 3D scanning - Laser Engraving Personalization.

￼

skasilov, 2024-10-31:
> image.png (view on web) <https://github.com/user-attachments/assets/752ee980-b247-4132-86bc-e5d1ce0feb6e>
> So have a look at your Library/Application Support/MeerK40t directory:
> image.png (view on web) <https://github.com/user-attachments/assets/70a2fbe5-f8d3-4053-af56-5f321b979c64>

This is very strange! 
I don't even have this folder there :-)))

Sincerely Sergey Kasilov
CEO/CTO
+31617349421

The Netherlands
Catharina van Rennesstraat 8,
2511AT, 's-Gravenhage
https://www.linkedin.com/in/sergey-kasilov
https://www.innerquill.com
https://www.etsy.com/shop/InnerQuill
https://www.instagram.com/innerquill.project

Unique tools for creativity and calligraphy.
3D printing - 3D scanning - Laser Engraving Personalization.

￼

￼

jpirnay, 2024-10-31:
I think I have covered it, so you need to wait for the compile of the new release.

skasilov, 2024-10-31:
> I think I have covered it, so you need to wait for the compile of the new release.

Perfect! Will be waiting with hope!

skasilov, 2024-11-01:
Problem solved with last version! Your are the BEST! Thank you!

jpirnay, 2024-11-01:
Great, good to hear

closed 2024-11-01 by jpirnay