meerk40t/meerk40t #2699: [Bug]:Version  0.9.5.2 Crash on Mac at startup, only the crash and splash screen are show.

manuel1414, 2024-10-31:
### Summary Description

[image]

### Crash logs

```shell
MeerK40t crash log. Version: 0.9.5200 on Darwin: Python 3.8.7: x86_64 - wxPython: 4.2.1 osx-cocoa (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "mk40t.py", line 11, in <module>
  File "meerk40t/main.py", line 198, in run
  File "meerk40t/main.py", line 230, in _exe
  File "meerk40t/kernel/kernel.py", line 1150, in __call__
  File "meerk40t/kernel/kernel.py", line 860, in set_kernel_lifecycle
  File "meerk40t/gui/plugin.py", line 410, in plugin
  File "meerk40t/kernel/kernel.py", line 2296, in console
  File "meerk40t/kernel/kernel.py", line 2342, in _console_parse
  File "meerk40t/kernel/functions.py", line 286, in inner
  File "meerk40t/gui/wxmeerk40t.py", line 710, in window_open
  File "meerk40t/gui/wxmeerk40t.py", line 699, in window_open
  File "meerk40t/kernel/context.py", line 419, in open_as
  File "meerk40t/gui/wxmmain.py", line 243, in __init__
  File "meerk40t/gui/mwindow.py", line 36, in __init__
  File "meerk40t/gui/themes.py", line 192, in set_window_colors
wx._core.wxAssertionError: C++ assertion ""IsOk()"" failed at /private/var/folders/ng/qbrmmqsx6990j5cfzz6t_z0h0000gn/T/pip-install-x53_dp_9/wxpython_8de19be4eb1141a98a5a5d6b51ee1eb4/ext/wxWidgets/src/osx/core/colour.cpp(219) in OSXGetNSColor(): invalid colour

Local variables:
[mk40t.py:11, in <module>]:
__name__ : __main__
__doc__ : None
__package__ : None
__loader__ : <class '_frozen_importlib.BuiltinImporter'>
__spec__ : None
__annotations__ : {}
__builtins__ : <module 'builtins' (built-in)>
__file__ : /Applications/MeerK40t.app/Contents/MacOS/mk40t.py
sys : <module 'sys' (built-in)>
pyimod03_importers : <module 'pyimod03_importers' from 'PyInstaller/loader/pyimod03_importers.py'>
os : <module 'os' from '/Applications/MeerK40t.app/Contents/MacOS/os.pyc'>
VIRTENV : VIRTUAL_ENV
python_path : ['/Applications/MeerK40t.app/Contents/MacOS/base_library.zip', '/Applications/MeerK40t.app/Contents/MacOS/lib-dynload', '/Applications/MeerK40t.app/Contents/MacOS']
pth : /Applications/MeerK40t.app/Contents/MacOS
NullWriter : <class '__main__.NullWriter'>
encodings : <module 'encodings' from '/Applications/MeerK40t.app/Contents/MacOS/base_library.zip/encodings/__init__.pyc'>
ctypes : <module 'ctypes' from '/Applications/MeerK40t.app/Contents/MacOS/ctypes/__init__.pyc'>
LibraryLoader : <class 'ctypes.LibraryLoader'>
DEFAULT_MODE : 4
_frozen_name : <function _frozen_name at 0x107c9c4c0>
PyInstallerImportError : <class '__main__.PyInstallerImportError'>
PyInstallerCDLL : <class '__main__.PyInstallerCDLL'>
PyInstallerPyDLL : <class '__main__.PyInstallerPyDLL'>
dyld : <module 'ctypes.macholib.dyld' from '/Applications/MeerK40t.app/Contents/MacOS/ctypes/macholib/dyld.pyc'>
d : /Applications/MeerK40t.app/Contents/MacOS/eggs
re : <module 're' from '/Applications/MeerK40t.app/Contents/MacOS/base_library.zip/re.pyc'>
multiprocessing : <module 'multiprocessing' from '/Applications/MeerK40t.app/Contents/MacOS/multiprocessing/__init__.pyc'>
spawn : <module 'multiprocessing.spawn' from '/Applications/MeerK40t.app/Contents/MacOS/multiprocessing/spawn.pyc'>
_args_from_interpreter_flags : <function _args_from_interpreter_flags at 0x10804fee0>
_freeze_support : <function _freeze_support at 0x107e64ca0>
forking : <module 'multiprocessing.popen_fork' from '/Applications/MeerK40t.app/Contents/MacOS/multiprocessing/popen_fork.pyc'>
spawning : <module 'multiprocessing.popen_spawn_posix' from '/Applications/MeerK40t.app/Contents/MacOS/multiprocessing/popen_spawn_posix.pyc'>
FrozenSupportMixIn : <class '__main__.FrozenSupportMixIn'>
_Popen : <class '__main__._Popen'>
_Spawning_Popen : <class '__main__._Spawning_Popen'>
tcldir : /Applications/MeerK40t.app/Contents/MacOS/tcl
tkdir : /Applications/MeerK40t.app/Contents/MacOS/tk
is_darwin : True
pathlib : <module 'pathlib' from '/Applications/MeerK40t.app/Contents/MacOS/pathlib.pyc'>
pkg_resources : <module 'pkg_resources' from '/Applications/MeerK40t.app/Contents/MacOS/pkg_resources/__init__.pyc'>
FrozenImporter : <class 'pyimod03_importers.FrozenImporter'>
SYS_PREFIX : /Applications/MeerK40t.app/Contents/MacOS
[... 160 lines cut ...]
width : 1728
height : 972
context : Context('/')
path : window/MeerK40t
parent : None
style : 541072960
args : ()
kwds : {}
[meerk40t/gui/themes.py:192, in set_window_colors]:
self : Service('themes', None)
win : MeerK40t(Context('/'), name="window/MeerK40t")
tp : {'win_bg': None, 'win_fg': None, 'button_bg': None, 'button_fg': None, 'text_bg': None, 'text_fg': None, 'list_bg': None, 'list_fg': None, 'label_bg': None, 'label_fg': None, 'highlight': None, 'inactive_bg': None, 'inactive_fg': None, 'pause_bg': None, 'pause_fg': wx.Colour(0, 0, 0, 255), 'start_bg': None, 'start_fg': None, 'start_bg_inactive': None, 'start_fg_focus': None, 'stop_bg_inactive': None, 'stop_bg': None, 'stop_fg': None, 'stop_fg_focus': None, 'arm_bg': None, 'arm_fg': None, 'arm_bg_inactive': None, 'arm_fg_focus': None}
```

### MeerK40t Version

0.9.5.2

### MeerK40t Type

Executable

### Your Operating System

Mac (before M1/Bigsur)

manuel1414, 2024-10-31:
Runing from source:

MeerK40t crash log. Version: 0.9.5200 src on Darwin: Python 3.12.0: x86_64 - wxPython: 4.2.1 osx-cocoa (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/idlelib/run.py", line 580, in runcode
    exec(code, self.locals)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t.py", line 11, in <module>
    sys.exit(main.run())
             ^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/main.py", line 198, in run
    _run = _exe(False, args)
           ^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/main.py", line 230, in _exe
    kernel()
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 1150, in __call__
    self.set_kernel_lifecycle(self, LIFECYCLE_KERNEL_SHUTDOWN)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 860, in set_kernel_lifecycle
    plugin(kernel, "mainloop")
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/plugin.py", line 410, in plugin
    kernel.console("window open MeerK40t\n")
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 2296, in console
    data_out = self._console_parse(command, channel=self._console_channel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 2342, in _console_parse
    data, remainder, input_type = funct(
                                  ^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/functions.py", line 286, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmeerk40t.py", line 710, in window_open
    window_open(None)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmeerk40t.py", line 699, in window_open
    path.open_as(window_uri, window_name, parent, *args)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/context.py", line 419, in open_as
    instance = open_object(self, instance_path, *args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmmain.py", line 243, in __init__
    super().__init__(int(width * 0.9), int(height * 0.9), *args, **kwds)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/mwindow.py", line 36, in __init__
    self.window_context.themes.set_window_colors(self)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/themes.py", line 192, in set_window_colors
    win.SetBackgroundColour(tp["win_bg"])
wx._core.wxAssertionError: C++ assertion ""IsOk()"" failed at /Users/robind/projects/bb2/dist-osx-py312/build/ext/wxWidgets/src/osx/core/colour.cpp(219) in OSXGetNSColor(): invalid colour

jpirnay, 2024-10-31:
It looks like you can run from source? Please fetch the latest code from GitHub and run it again, and tell me what message is given on the console. I would be grateful for your meerk40t.cfg file, too, to be able to check what's going on.
Are you running on proper Apple Hardware or on a hackintosh style system?

jpirnay, 2024-10-31:
Duplicate to issue #2699 - will continue discussion there
In a nutshell, should be fixed

closed 2024-10-31 by jpirnay

manuel1414, 2024-10-31:
This is the log in Terminal:

iMac-de-Manuel:~ Laser$ python3 /Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t.py
OpenCV is not installed. Disabling Camera. Install with:
	pip install opencv-python-headless
MeerK40t crash log. Version: 0.9.5200 src on Darwin: Python 3.12.0: x86_64 - wxPython: 4.2.1 osx-cocoa (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t.py", line 11, in <module>
    sys.exit(main.run())
             ^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/main.py", line 198, in run
    _run = _exe(False, args)
           ^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/main.py", line 230, in _exe
    kernel()
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 1150, in __call__
    self.set_kernel_lifecycle(self, LIFECYCLE_KERNEL_SHUTDOWN)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 860, in set_kernel_lifecycle
    plugin(kernel, "mainloop")
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/plugin.py", line 410, in plugin
    kernel.console("window open MeerK40t\n")
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 2296, in console
    data_out = self._console_parse(command, channel=self._console_channel)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/kernel.py", line 2342, in _console_parse
    data, remainder, input_type = funct(
                                  ^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/functions.py", line 286, in inner
    returned = func(command=command, channel=channel, **ik, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmeerk40t.py", line 710, in window_open
    window_open(None)
[MeerK40t.txt](https://github.com/user-attachments/files/17592698/MeerK40t.txt)

  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmeerk40t.py", line 699, in window_open
    path.open_as(window_uri, window_name, parent, *args)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/kernel/context.py", line 419, in open_as
    instance = open_object(self, instance_path, *args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/wxmmain.py", line 243, in __init__
    super().__init__(int(width * 0.9), int(height * 0.9), *args, **kwds)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/mwindow.py", line 36, in __init__
    self.window_context.themes.set_window_colors(self)
  File "/Users/Laser/Downloads/meerk40t-0.9.5200/meerk40t/gui/themes.py", line 192, in set_window_colors
    win.SetBackgroundColour(tp["win_bg"])
wx._core.wxAssertionError: C++ assertion ""IsOk()"" failed at /Users/robind/projects/bb2/dist-osx-py312/build/ext/wxWidgets/src/osx/core/colour.cpp(219) in OSXGetNSColor(): invalid colour

Abort trap: 6
iMac-de-Manuel:~ Laser$

manuel1414, 2024-10-31:
[MeerK40t.txt](https://github.com/user-attachments/files/17592706/MeerK40t.txt)