meerk40t/meerk40t #2729: [Bug] 0.9.5.3 Dark Mode - default=system, splash screen

Sophist-UK, 2024-11-10:
1. It was a surprise to get 0.9.5.3 start first time with different colours from before. I think that the Dark mode switch should not default to "System" for first run for existing users, but instead to "Light". For first time users, I guess "System" has more meaning.

2. Splash screen should IMO always be shown as Light theme - because otherwise it looks weird as a negative image.

Sophist-UK, 2024-11-10:
Update - splash screen showed correctly on my (non-production) laptop, but was still cropped to only part of the image on my Surface Pro 3.

jpirnay, 2024-11-12:
The cropped image is part of issue #2728
Everything is else is a feature request

Sophist-UK, 2024-11-12:
I disagree with the default for existing users resulting in a sudden switch to dark mode being a feature request. It was a bad feature enhancement that causes a significant hiccup to the user experience.

jpirnay, 2024-11-12:
So we agree to disagree

Sophist-UK, 2024-11-12:
Splash screen works with scaling of 150% (but not 165%) as suggested in #2731. Not sure what #2728 has to do with this, though.

jpirnay, 2024-11-15:
Addressed in 
- https://github.com/meerk40t/meerk40t/commit/14db72f5dcf54954cdd1c3d55c52b2763746e5bb
- https://github.com/meerk40t/meerk40t/commit/c27cf9eb777529a2dbcfefe6d89a5e3e36690cbd

closed 2024-11-15 by jpirnay