meerk40t/meerk40t #2735: Suggestion: UI toolbar change

Sophist-UK, 2024-11-12:
# Current
### Project
[image]
### Modify
[image]
### Config
[image]

The difficulty is that from a workflow perspective, the `Modify` primary toolbar really goes with the secondary toolbar, and it probably doesn't make sense to have the same secondary toolbar for all three primary toolbars.

However I am unclear what the alternative might be:
* Move the `Modify` primary toolbar to a 2nd secondary toolbar; or
* Move the secondary toolbar to be a primary `Drawing` toolbar; or
* ...

Perhaps it is worth soliciting views from various users...

jpirnay, 2025-01-30:
There is now another (experimental) toolbar with the "modify" tab content

Sophist-UK, 2025-01-30:
I apologise if I didn't explain clearly but I was actually suggesting adding more tabs, and not moving an existing tab into a pane instead.

The issue with panes is that showing / hiding them is messy, and leaving them showing permanently uses valuable screen real-estate. So personally I quite like the tab approach which with a single click changes the icons shown depending on the type of actions you are trying to do. Since workflow often involves doing several of the same types of action, this seems to make a lot of sense to me.

Perhaps the most flexible approach would be to put all the tabs into panes that you can show permanently if you wish, and then to allow the user to define which panes can then be displayed as tabs and switched between.