meerk40t/meerk40t #2741: [Bug (minor)] 0.9.5.3: Burn optimised sequence is sub-optimal

Sophist-UK, 2024-11-22:
I have no idea why but:

1. Burn sequence of 2 images and 2 raster elements is sub-optimal on first element; 
2. Burn sequence is not always the same

I suspect that the starting point for the burn is not being initialised properly.

[image]

In the above image you can see that it goes first to the stars image and only does the raster words at the top at the end - I would expect it to start with the words at the top and then move downwards to the next element.

I didn't capture the simulation as an image (and haven't managed to recreate it), but having done an actual burn with the above sequence, when I opened the simulation window the second lower set of words was shown as burning first.

Sophist-UK, 2024-11-22:
Here is an example of the second issue.

I started a burn for two of these, then deleted one and opened simulation.

What is even weirder is that not all the burn paths for the deleted item are in the simulation - only some of them.
### Before delete:
[image]
### After delete:
[image]
### Simulation:
[image]

jpirnay, 2024-12-03:
Acknowledged

Sophist-UK, 2024-12-04:
These sorts of intermittent problems can be very difficult to track down. IMO it is better to prevent code from having incorrect input and raising an exception (which shows exactly what was wrong and why) than to allow it to happen and give false output and then have to try to track down where it is going wrong. That is where strong typing comes in - be explicit about what you expect to be passed into a method and let there be show-stopping exceptions when it doesn't happen. (David was a talented coder, but in this respect he has left a legacy of unreliable code.)

jpirnay, 2024-12-17:
On the first matter - do these operations contain by any chance a definite coolant setting? This would cause this behaviour as the operations would not be able to merge. E.g.
- With coolant setting [image]
- Without 
[image]

Sophist-UK, 2024-12-17:
No - I have no coolant control and haven't ever set this to my knowledge. But I can check the actual file.

closed 2025-07-07 by jpirnay