meerk40t/meerk40t #2747: Suggestion: An olptimisation for greedy optimisation

Sophist-UK, 2024-11-30:
I burn for leathwork which often has hundreds of small circular-shaped paths for vector engraves.

At present greedy optimisation finds the next circular path by optimising over each of 4 segments of each of the circular paths.

My suggestion is that when optimising you check the maximum of x & y sizes from the bounding box and if it is smaller than (say) 5mm you use e.g. the centre point of the bounding box for the distance rather than the end points of each of the segments in the path and then if this turns out to be the closest you do a small iteration over the start and end points to find the closest segment. 

I don't think this would work for larger paths because of the variations of sizes and shapes.

P.S. I cannot remember from when I wrote some of the optimisations whether we check both start and end points of each segment, but clearly (when each path is already a simple path and not a compound one) the start of the next segment is the same as the end of the current segment. So if we are checking distances of both start and end points we only need to calculate the start point distances of each segment, plus for a non-closed path the end point of the last segment and then for a non-closed path choose which way to burn from that point.

jpirnay, 2024-12-03:
Thank you for your suggestion. In all honesty the complete optimisation logic requires an eventual overhaul. It has grown in complexity and deteriorated in performance. 
Just one example: the candidate search for possible start / end points of closed paths provide *all* segments of that path - while this is 'correct' in the sense of 'a closed path does not have a defined start/end point, so any point could be a good / a bad choice' this leads to a myriad of required comparisons consuming often any time savings on the device at the time of calculation.
This needs to go on the backlog but definitely holds some merit.

Sophist-UK, 2024-12-04:
For when you do revisit this, a few more ideas as a braindump for optimising greedy optimisation:

* Break path elements into individual simple paths (as present) - there is little point in trying to optimise using compound paths. 

* That said, there may be some merit in checking whether a non-closed compound path can be simplified by joining two simple paths because they have a common end-point (which would need to be done after the existing check for whether a simple open path is effectively closed and treating it as closed if the start and end-points are coincident).

* For elements within the same operation or with `merge operations` selected, it might be useful to do a one-time NUMPY sort of a list of start and end points for non-closed paths to see if they have coincident end points and can be "merged" into a simple path (same operation) or some kind of merged path (different operation). OTOH even though this is a one-time operation, this might be a bit too expensive for the savings it might generate.

* For each path, you also want to pre-analyse whether it can be burned in segments i.e. either it is a closed path or you have "Burn complete subpaths" not checked.

* If you are doing burn inner first grouped, then analyse what is inside what and create groups of paths for optimisation. Once you start burning a group, then you only need to optimise within that group. If you have nested groups then you need to focus on the most inner group containing the previous burn. You can also ignore the outer cut until everything inside has been burned.

* Try to find a way to iterate cheaply to produce a subset of objects to consider burning next based on simple bounding box comparisons rather than Pythagoras. i.e. any path whose nearest corner is further away in either X or Y than another paths most distant corner, we know we can ignore this path. It should be possible to use NUMPY to make these calculations and give a subset of paths to consider in detail, and it might be possible to use NUMPY to then calculate and sort the accurate pythagorus distances for the path segments within that subset.

* If the previous burn ends reasonably close to the point used for the previous subsetting, then the subset of paths considered for that burn is probably STILL a pretty good candidate for the next burn, so perhaps it can be reused. The difficulty is to know when to use it and when to calculate the subset again.

So at this point you have a list of candidate paths that hopefully is much much smaller than the full list. And only now do you look inside these paths to consider path segments - and you use the start and end points of each segment onto if you have predetermined that segments can be burned in any order and not paths that need to be burned from start to finish.

* Now you repeat the two stage optimisation (first using x & y distances and secondly using Pythagorean distances) using start and end points on this subset of paths (ideally using NUMPY again) until you identify the paths or path segments that are closest (within a margin of error).

* Now it is possible that you have two or more paths/path segments that have (approximately) co-incident end-points. You want the choice between them to be deterministic so that a burn is exactly repeatable. Now, clearly a short burn (in terms of distance between start and end points) which allows you to re-use the subset of paths you previously created is going to result in less compute power than an alternative path that takes you away from the locale you are in, so you should see if there is a closed path in the list and do that first (in the forward direction) - and then if there are any points left in the short-list of a short-list you can keep using this short-list and not recalculate it. If there are no closed paths, then personally I would select the shortest path (in terms of distance between start and end points) and not do anything more sophisticated, but this may not be the best algorithm. 

* Then you burn all continuous segments in the path chosen until you reach the open end-point or exhaust the segments in the path.

This algorithm is NOT that much different from the current one except how it stores stuff when optimising (which is more hierarchical than the current algorithm which essentially flattens every path into path segments right at the start). Ideally you would hold all this stuff in NUMPY objects to avoid keeping converting them from non-NUMPY objects.