meerk40t/meerk40t #2752: [Bug]: Poor performance for some files

pkochubey, 2024-12-05:
### Summary Description

[image]
You can use this file for testing. On last version trace works very slow. 

If in **is_inside** function replace **return other_code** on **sb_code** all works very fast. I think we have some problem in function**other_code**

### Additional Details

perf_counter beetwen sb_code and other_code:

Tolerance: 39.37007874015749, sb=True in 0.000s, other=True in 4.955s
Tolerance: 39.37007874015749, sb=True in 0.001s, other=True in 4.052s

### MeerK40t Version

latest from git

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

pkochubey, 2024-12-06:
@jaredly 
need use more big(100) **distance** for arc in this place https://github.com/meerk40t/meerk40t/blob/9715673dc65d9f9cdb231d655303fe52685ce59a/meerk40t/core/cutplan.py#L1144 :

```
points = np.array(list((p.real, p.imag) for p in geom1.as_equal_interpolated_points(distance = 10)))
vertices = np.array(list((p.real, p.imag) for p in geom2.as_equal_interpolated_points(distance = 10)))
```

maybe this is value shoudbe variable for different object.

jpirnay, 2024-12-07:
The scanbeam algorithm you are referring too has issues, so it's not consistently providing the right answer despite being fast...
So better slow than wrong.

closed 2024-12-07 by jpirnay

jpirnay, 2024-12-08:
I have made a couple of adjustments, check again with your file - work okay on my machine