meerk40t/meerk40t #2766: [Suggestion] Make several executable versions with different Python versions

Sophist-UK, 2024-12-14:
Later versions of Python are substantially faster than earlier versions.

We need to retain executable builds using older versions of Python to e.g. retain support for older versions of Windows/Linux/Mac etc.

My suggestion is to create multiple e.g. Windows executable versions so that the Windows 7 version uses the latest Python build that supports that, ditto for Windows 10 and Windows 11.

jpirnay, 2024-12-19:
While I understand and support the proposal, it is actually rather difficult to execute, some examples 
- the major acceleration lever for dithering images is the use of numba, the jit compiler. This is not available under python 3.13 
- a couple of other libraries like pypotrace-windows require a compilation stage that needs very specific c compiler settings 

Github actions manages to create more non-working versions than I care to count
This compilation nightmare requires obviously more insights/competencies that I am able to provide. So any help by whoever would be welcome

Sophist-UK, 2024-12-19:
Clearly no numba for 3.13 (yet at least) is a show stopper for that version.

And sorry, but I am not the "help by whoever" - until I get a guarantee that the Discord server is safe from abuse I won't be rejoining the development team (and whilst paiin is a moderator I will never believe that to be the case).

jpirnay, 2024-12-20:
Let me be crystal clear:

- All contributions to the MeerK40t project are welcome - regardless from where they are coming
- Any conditions imposed to contribute will be ignored and will effectively prevent them from being taken onboard

So it's your as anyone elses choice whether that individual will feel comfortable contributing and we will accept that decision, but we won't change the rules

Sophist-UK, 2024-12-20:
I am NOT imposing conditions, I am simply saying that I am not contributing code whilst I believe that the development team Discord channel can still be used by one developer to abuse another (which being a moderator they can do with impunity).

If the rules now say that abuse is not tolerated under any circumstances then please let me know that this is now the case. If the rules say that abuse is tolerated (even by omission) then that says a LOT about the management of the Discord server,

jpirnay, 2024-12-20:
Stop using github to fight your private vendetta - if you feel unfairly treated by me or anyone else it's your perogative to turn your back to this project and anyone involved with it. I am getting tired to read in every other issue how you feel about the past.
I am not saying that the behaviour you witnessed has been correct or abusive nor am I saying your actions have been flawless / flawed. I am saying though that this is not a discussion forum - if you feel the need to discuss then there are other communication channels you can choose.

Sophist-UK, 2024-12-20:
I do not say this "in every other issue". But to be honest I thought you were a better person than to turn a blind eye and do nothing to prevent it happening again. 

But, if you are asking people to join the development community then they need to know that it is not a safe space and to know the risks they are taking.