meerk40t/meerk40t #2767: [Bug] 0.9.5.3: Undo issues

Sophist-UK, 2024-12-14:
Two issues:

1. Undo is unreliable - sometimes one undo actually undoes several previous changes, sometimes redo does nothing.

2. Can we please save the status of open / closed nodes in the tree and possible selected elements when we save an undo state and restore it on redo. (At present the tree is refreshed and all nodes are closed after an undo or redo.)

jpirnay, 2025-01-07:
Acknowledged

jpirnay, 2025-01-14:
Should be much better after PR #2784

Sophist-UK, 2025-01-18:
Just tried 0.9.7b3 and on a limited one time set of undos it seems reliable and fast. 👍

jpirnay, 2025-01-24:
Unfortunately the undo stack can still be corrupted by jumping back to a state starting from there again and then undo again to the first undo marker. Just fixing it...

jpirnay, 2025-01-24:
[image]

jpirnay, 2025-01-30:
Fixed in 0.9.7b4

closed 2025-01-30 by jpirnay