meerk40t/meerk40t #2770: [Bug - Major] 0.9.7b1 will not connect to m2nano - 0.9.5 does

Sophist-UK, 2024-12-17:
0.9.7b1:
```
    Finding devices.
    No backend available
    PyUsb detected no backend LibUSB driver.
    
    USB connection did not exist.
    Using CH341 Driver to connect.
    Attempting connection to USB.
    No CH341 Devices detected.
    
    Connection failed.
```

0.9.5:
```
  Using LibUSB to connect.
    Attempting connection to USB.
    
    
    Finding devices.
    1 device(s) were found.
    K40 device detected:
    DEVICE ID 1a86:5512 on Bus 000 Address 001 =================
    	 bLength                :   0x12 (18 bytes)
    	 bDescriptorType        :    0x1 Device
    	 bcdUSB                 :  0x110 USB 1.1
    	 bDeviceClass           :   0xff Vendor-specific
    	 bDeviceSubClass        :    0x0
    	 bDeviceProtocol        :    0x2
    	 bMaxPacketSize0        :    0x8 (8 bytes)
    	 idVendor               : 0x1a86
    	 idProduct              : 0x5512
    	 bcdDevice              :  0x304 Device 3.04
    	 iManufacturer          :    0x0 
    	 iProduct               :    0x0 
    	 iSerialNumber          :    0x0 
    	 bNumConfigurations     :    0x1
    	  CONFIGURATION 1: 100 mA ==================================
    	   bLength              :    0x9 (9 bytes)
    	   bDescriptorType      :    0x2 Configuration
    	   wTotalLength         :   0x27 (39 bytes)
    	   bNumInterfaces       :    0x1
    	   bConfigurationValue  :    0x1
    	   iConfiguration       :    0x0 
    	   bmAttributes         :   0x80 Bus Powered
    	   bMaxPower            :   0x32 (100 mA)
    	    INTERFACE 0: Vendor Specific ===========================
    	     bLength            :    0x9 (9 bytes)
    	     bDescriptorType    :    0x4 Interface
    	     bInterfaceNumber   :    0x0
    	     bAlternateSetting  :    0x0
    	     bNumEndpoints      :    0x3
    	     bInterfaceClass    :   0xff Vendor Specific
    	     bInterfaceSubClass :    0x1
    	     bInterfaceProtocol :    0x2
    	     iInterface         :    0x0 
    	      ENDPOINT 0x82: Bulk IN ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x82 IN
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x2: Bulk OUT ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :    0x2 OUT
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x81: Interrupt IN ==========================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x81 IN
    	       bmAttributes     :    0x3 Interrupt
    	       wMaxPacketSize   :    0x8 (8 bytes)
    	       bInterval        :    0x1
    Config Set
    Config Set: Success
    Getting Active Config
    Active Config: Success.
    Kernel detach: Not Implemented.
    Attempting to claim interface.
    Interface claim: Success
    USB Connected.
    Sending CH341 mode change to EPP1.9.
    CH341 mode change to EPP1.9: Success.
    Device Connected.
    
    CH341 Chip Version: 50
    Driver Detected: LibUsb
    Device Connected.
```

jpirnay, 2024-12-17:
True for the version with all bells and whistles. Removed the culprit exe. The compilation process under github actions is a nightmare.

closed 2024-12-17 by jpirnay

Sophist-UK, 2024-12-17:
I did notice that the executable had doubled in size, but no idea why - I just downloaded the one that looked like previous versions (the Meerk40t.exe) and ignored the "minimal one" because I didn't want (what I thought from the name was) a cut down version.