meerk40t/meerk40t #2772: [Bug]: In Mac the textfield background appears in black making imposible see the characters inside it.

manuel1414, 2024-12-18:
### Summary Description

When you click on a textfield, the background turs or is in black, making imposible see de text inside of it. 
[image]

### Additional Details

All versions of Meerk40t from 8.1 to 9.53.

### MeerK40t Version

All version of Meerk40t

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Mac (before M1/Bigsur)

jpirnay, 2024-12-19:
We are using it on multiple macs without isdues, so I need additional information please:
- which MacOS version
- what is the major UI theme: dark/light
- which UI themechave you chosen in meerk40t: light/system default

jpirnay, 2025-02-25:
Not getting any feedback... Will close it it after another 5 days

closed 2025-03-06 by jpirnay