meerk40t/meerk40t #2793: [Bug] 0.9.7b3: Merge Operations seems to have ceased working

Sophist-UK, 2025-01-28:
[image]

jpirnay, 2025-01-29:
Additional input from discord: 

I mostly burn leather patterns which typically consist of several separate component pieces of leather each of which has vector engraved stitch holes surrounded by an edge cut. ANd I burn these with Merge Operations, Burn Inside First and Group Burns together so that the stitchholes of one piece are burned first and then the edge of the same piece before another piece is burned.

I notice that recent versions of MK (or at least the most recent beta) it appears that Merge Operations has stopped working so it burns first the stitch holes of every piece and then the cut edges of every piece. (If Merge Operations was working but something else had stopped working, I think it would have mixed the engraves andthe cuts in some way, just not the way they should be mixed - but since it does all of the Engraves and then all of the Cuts, it appears that Merge Operations is the culprit.

jpirnay, 2025-01-29:
This is due to unwanted side effect of coolant recognition - at least one of the engrave / cut operation has an explicit "set coolant" to either on or off:
[image]

If you would set both to "No changes" then the merge operation should work again
[image]

Recognized and fixed in latest main

jpirnay, 2025-01-29:
- To avoid this side effect you need to fall back to 0.9.5.3
- Devices already have that setting, look at the device config

jpirnay, 2025-01-30:
Fixed in 0.9.7b4

closed 2025-01-30 by jpirnay