meerk40t/meerk40t #2816: come capovolgere i numeri sul piano di lavoro (turn up side down number)

Alecs70, 2025-02-11:
Non so come mai i numeri sull'area di lavoro risultano rovesciati. Come si possono mettere dritti?
Grazie

[image]

jpirnay, 2025-02-12:
Wow, that's fascinating and the very first time I am seeing this - the complete scene is upside down!
So: the usual questionnaire:
a) Click on Help - About Meerk40t (Aiuto - Informazione sul MeerK40t) and copy the content of the information and component tabs into the clipboard and paste it here
[image]
b) Any peculiar monitor setup you are working with (rotated screens)?
c) What are the definitions of your laser-working area, click on the printer properties, eg for grbl:
[image]

Alecs70, 2025-02-12:
[image]

[image]

[image]

jpirnay, 2025-02-12:
Hmm, no indicaton of somehing fully amiss - can you please start meerk40t from the command line via ``meerk40t.exe -X`` to nuke the settings file. Is the phenomenon still in place then?

Alecs70, 2025-02-12:
Scusa non capisco cosa intendi per:  avviare meerk40t dalla riga di comando tramite meerk40t.exe -X

Come devo fare?

jpirnay, 2025-02-13:
Open a terminal window (prompt dei comandi ?), change to the directory, where you have put the meerk40t.exe file (something like 'cd \tools\meerk40t') and enter 'meerk40t -X'

Alecs70, 2025-02-13:
Grazie jpirnay, pare essersi risolo con il tuo contributo.
Ora devo solo invertire la scala dell'asse Y

[image]