meerk40t/meerk40t #2822: [Bug]: Crash after opening simulation window

bwmalone, 2025-02-23:
### Summary Description

Program crashes after loading the Simulation window.
-  
Linux Mint 21.3 Cinnamon - last message to terminal:
(MeerK40t-Linux-Ubuntu-22.04:319017): Gtk-WARNING **: 20:58:37.501: Negative content width -1 (allocation 25, extents 13x13) while allocating gadget (node button, owner GtkButton)

(MeerK40t-Linux-Ubuntu-22.04:319017): Gtk-WARNING **: 20:58:37.741: Negative content width -1 (allocation 25, extents 13x13) while allocating gadget (node button, owner GtkButton)
Segmentation fault (core dumped)

Happy to help isolate if needed. 

### Crash logs

```shell

```

### MeerK40t Version

v0.9.7010

### MeerK40t Type

Executable

### Your Operating System

Linux

jpirnay, 2025-02-23:
We had this some time ago and I am not able to recreate the issue - so I would need additional information, please: can you describe the scenario (always, only for certain files / devices) and share some culprit files

bwmalone, 2025-02-23:
I get the suspicion it may be a version of a dependent lib.

It happens even on a new file immediately after opening the app
 usually within 2 seconds of opening the window. Happens nearly every time
and doesn't seem to matter what file I open. Have only been able to use the
feature 1 time.

Will take a video when I get home later.

jpirnay, 2025-02-23:
Are you able to run from source? Then please use the branch [fixes](https://github.com/meerk40t/meerk40t/tree/fixes) and share the console output. I would recommend to run meerk40t via ``python3 meerk40t.py 2>/dev/null`` to suppress the annoying wxPython debug messages.

bwmalone, 2025-02-23:
I don't have much experience with python, but I'll give it a shot later.
Should be able to.

bwmalone, 2025-02-24:
Took a minute to get all the dependencies figured out...

But here it is:

python3 meerk40t.py >/dev/null
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown
18:47:07: Debug: ClientToScreen cannot work when toplevel window is not
shown

(meerk40t.py:336812): Gtk-WARNING **: 18:47:09.103: Negative content width
-1 (allocation 25, extents 13x13) while allocating gadget (node button,
owner GtkButton)

(meerk40t.py:336812): Gtk-WARNING **: 18:47:09.298: Negative content width
-1 (allocation 25, extents 13x13) while allocating gadget (node button,
owner GtkButton)
Segmentation fault (core dumped)

bwmalone, 2025-02-24:
> Also not sure if relevant but padding around some objects looks
> non-existent. If there is a modification to an object width that is for
> whatever reason 0:

Last message tonight I promise.

After playing with it a Lot... Sometimes when it crashes the Send to Laser
button disappears after the window renders. If the button disappears... It
definitely crashed within 2 seconds

jpirnay, 2025-02-24:
Thank you for your efforts - unfortunately you suppressed accidentally the relevant output. The command to use is ``python3 meerk40t.py 2>/dev/null`` - the 2 before the greater sign is relevant!
So could you please rerun the tests? Thx

bwmalone, 2025-02-24:
I hadn't noticed that! Will rerun when I get a break from work.

bwmalone, 2025-02-24:
Unfortunately that provided even less information:
[image: image.png]

bwmalone, 2025-02-24:
I kept futzing with it and got a few things that might help:

*Saw this one time just before the seg fault:*
(meerk40t.py:337255): Gtk-CRITICAL **: 18:57:09.245:
gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar

*Saw this one time app locked up on opening the simulate window, repeating
the errors in this stack indefinitely until I killed it.*

(meerk40t.py:355646): Pango-CRITICAL **: 11:08:20.440:
pango_context_get_matrix: assertion 'PANGO_IS_CONTEXT (context)' failed

(meerk40t.py:355646): GLib-GObject-WARNING **: 11:08:20.440: invalid
unclassed pointer in cast to 'GObject'

(meerk40t.py:355646): GLib-GObject-CRITICAL **: 11:08:20.440:
g_object_get_qdata: assertion 'G_IS_OBJECT (object)' failed

(meerk40t.py:355646): Pango-CRITICAL **: 11:08:20.440:
pango_context_get_matrix: assertion 'PANGO_IS_CONTEXT (context)' failed

(meerk40t.py:355646): GLib-GObject-WARNING **: 11:08:20.440: invalid
unclassed pointer in cast to 'GObject'

(meerk40t.py:355646): GLib-GObject-CRITICAL **: 11:08:20.440:
g_object_get_qdata: assertion 'G_IS_OBJECT (object)' failed

(meerk40t.py:355646): GLib-GObject-WARNING **: 11:08:20.440: invalid
unclassed pointer in cast to 'GObject'

jpirnay, 2025-02-24:
There are no debug messages at all... Did you use the ``fixes`` branch as indicated above?

bwmalone, 2025-02-24:
Pretty sure... I'll nuke the local folder and re download to be sure

bwmalone, 2025-02-24:
When I used the GH command line, it grabbed the main branch... I had to
download the zip to get the fixes.

*I think this is what you are looking for?*

Init done: 1.981999957934022e-06
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Plan called : 0.04491512000095099 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.04495985101675615
Startup: 0.04516723699634895
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Init done: 8.089991752058268e-07
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Plan called : 0.01098995201755315 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.011032378999516368
Startup: 0.011245364003116265
Plan called : 0.012107916991226375 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.012135415017837659
Startup: 0.012295555992750451
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Init done: 7.179914973676205e-07
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Plan called : 0.014427139016333967 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.014490638015558943
Startup: 0.014788836007937789
Plan called : 0.015734646993223578 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.01576201998977922
Startup: 0.01594468700932339
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Init done: 8.179922588169575e-07
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Plan called : 0.015889915986917913 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.01594888500403613
Startup: 0.016206008003791794
Plan called : 0.017213240993442014 (Pane: (666, 504), Sizer: (706, 504),
Window: (706, 718))
Refresh simulated: 0.01725218800129369
Startup: 0.017486970988102257
Manually forwarding the size: (706, 718)
Now pane has: (666, 504)
Segmentation fault (core dumped)

bwmalone, 2025-02-25:
> That's what I was looking for, can you please provide some context what
> you did? I am counting 4 inits = 4 times opening the simulation window?!

Yes I was only trying to reproduce. That time took four tries to open the
window. Sometimes it's 1st time sometimes it's 5. Happens more often if I
have no elements or more than 50

jpirnay, 2025-02-25:
I was finally able to erratically reproduce the issue, after some changes to the progress update mechanism (which isn't properly running under Linux anyway due to some wxWidgets constraints) I was no longer able to reproduce the issue. But: the absence of evidence is not necessarily the evidence of absence in this case. So your mileage may vary. Please provide some feedback whether you still encounter the reported issue (and no there aren't any debug statements any more to track and share as they were irrelevant to the case)

bwmalone, 2025-02-25:
Thanks - I assume on the fixes branch for testing?

I can usually repro it within 30 seconds - will update and try again this
afternoon.

bwmalone, 2025-02-26:
I saw the PR on main... Still crashing after updating

jpirnay, 2025-02-26:
That's unfortunate - could you join the discord server to discuss this in a wider group as I am running out of ideas.

bwmalone, 2025-02-26:
Will do.

I need to join anyway, as I need help to figure out how to replicate one of
the features in lightburn. Otherwise I'm going to have to go back as it's a
must-have

jpirnay, 2025-02-27:
The update routine did not run in the main thread context - fixed thanks to a great analysis by tiger

closed 2025-02-27 by jpirnay