meerk40t/meerk40t #2843: The X axis is not moving when trying to engrave on a glass using a laser rotary roller

1OldWoman, 2025-03-28:
When trying to put a design on a glass, the laser only moves on the Y-axis not the X- axis so it only creates a line and not the design.  The instructions on the rotary engraver says to put $10=80 in the box below the progress, which I've done and have tried it without that code, and it still does not move the X-axis.  

Is there another setting?

joerlane, 2025-03-28:
Is the rotary also spinning (slowly) along with the laser moving in the Y axis? Which orientation on the glass is the line being drawn? 

My first thought is that the rotary engraver needs to be turned 90 degrees with respect to the laser. Typically the laser should be drawing in the one moving axis on the glass with lines top to bottom (mouth to base), and the rotary slowly turns the glass during that process to complete the design.

1OldWoman, 2025-03-28:
Ok, Let me try that

1OldWoman, 2025-03-28:
That was it (duh on my part)!!! Thank you!

1OldWoman, 2025-03-29:
Yes! I totally had the rotary in the wrong direction. It’s working fine now!  Thank youSent from my iPhone

joerlane, 2025-03-29:
No problem; glad to help you get it working!

closed 2025-03-29 by joerlane

1OldWoman, 2025-03-29:
Thank you!!Sent from my iPhone