meerk40t/meerk40t #2845: [Bug]: Auto Sort not sorting ''Fill'' elements by color properly

klifinpol, 2025-04-03:
### Summary Description

[image]

When Importing an .SVG file into the program all of the ''Fill'' objects/elements get sorted into the same operation regardless of their color.

### Additional Details

I'm unsure whether this is as intended or not, however it's rather annoying to deal with... I wish there was a way to automatically seperate the fill SVG elements by color instead of having to manually asign them to seperate operations...  

For more context I work with a ton of different acrylics (of varying color) and in order to keep the designs of consistent size so that in the producst fit together in the end (size wise) I make material changes by enabling/disabling operations of different colors and cutting only the needed parts of the designs out...  So it really helps that the elements get grouped together into operations by color.

However when it comes to rasterisation I always run into the same issue where I have to manually sort all the Fill objects onto seperate operations.... A lot of the times I'll make copies of the same design and if I have different materials both neededing rasterisation the new copies will all auto sort into that one operation even though they are already assigned to a different operation... 

It's rather annoying to deal with when there is a very convinent autosort system already for other elements...
If I'm missing somthing or there perhaps is already a setting which autosorts Fill elements by color please let me know

### Crash logs

```shell

```

### MeerK40t Version

0.9.7020

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2025-04-04:
This isn't a bug, any object with a solid fill is assigned to a single raster operation, on the other habd any 'hollow' (ie fill none)  object will be assigned to a different operation based on their color.
What is your usecase for different fills?

jpirnay, 2025-04-04:
Sorry, just read your message properly, still a feature request...

klifinpol, 2025-04-04:
Fair enough, I hope I was able to explain my use case well enough to understand... I know it's a neiche case but it's very usefull if people intend to do material changes between cuts as it streamlines the enitre process and allows you to keep the sizes consistent throught different materials. If there's any further information which I can provide please let me know, I'd gladly like to help! The program you guys made is extrordinary!

jpirnay, 2025-04-17:
Check branch: fill_classify

[image]

klifinpol, 2025-04-22:
Thank you SO SO much!!! <3

closed 2025-06-16 by jpirnay