meerk40t/meerk40t #2850: k40 laser power

enricodeiure, 2025-04-14:
### Summary Description

I have a k40 laser with a nano m2 board, how should I adjust the laser power? 
If I set it to 1000, the laser goes to maximum power even if the potentiometer is set to half power

### Crash logs

```shell

```

### MeerK40t Version

last

### MeerK40t Type

Executable

### Your Operating System

Windows

Sophist-UK, 2025-04-14:
1. If you haven't already got one, fit a mA meter.

2. Press the test button and adjust it until the mA meter reads 12mA (which is full power for your K40).

3. Then in Meerk40t use the Power context menu for each operation to set the power relative to 100% you have set on the hardware.

Sophist-UK, 2025-04-14:
P.S. This is an issue of how to use the K40 and MK and NOT a bug (i.e. not an error in the code).

tiger12506, 2025-04-14:
`If I set it to 1000, the laser goes to maximum power even if the potentiometer is set to half power`

This is not actually possible. The potentiometer is a hardware limit in the power supply. So at most in that situation, the laser is putting out 100.0% of the half power that you have the potentiometer set at.