meerk40t/meerk40t #2853: [Bug]: Hatch parameters not used

nathan-stewart, 2025-04-19:
### Summary Description

Hatch Parameters appear to be in Config->Effects rather than an operation parameter, but the hatch distance and angle values have no effect and only 1mm 0 degrees are used. 

### Additional Details

[hatch-test.zip](https://github.com/user-attachments/files/19821305/hatch-test.zip)

### Crash logs

```shell

```

### MeerK40t Version

Latest version on github: MeerK40t v0.9.4000 (0.9.4: Many Fixes - Vector Fonts, GRBL, Material Manager)

### MeerK40t Type

Executable

### Your Operating System

Linux

jpirnay, 2025-04-19:
Are you really using 0.9.4? The latest version is 0.9.73 - there were a couple of changes around the hatch treatment so you might want to check this version (amongst many more reasons).
The file you shared does indeed only have 1mm / 0deg as parameters, but changing them to some arbitrary values doe show the desired effect.
[image]

nathan-stewart, 2025-04-19:
Same behavior - the test still creates them with 1mm spacing,  If I remove effect then add it I can chose other spacings (0.1mm) but not the value set in the config. Also removing and applying the effect only works for one of my operations the others do not work.

jpirnay, 2025-04-19:
The test will always create them with 1mm/0deg. You need to adjust them yourself. May be a useful thing to extend the test parameters.

nathan-stewart, 2025-04-19:
But how do you adjust them once created? I can't figure out how to do that without removing the effect and applying a new one which only works once and often causes crashes (reported)

jpirnay, 2025-04-20:
Hatches can be applied to...
- an element individually. Advantage: you see what you will get immediately on your design canvas

[image]

- to an operation as a whole, all elements assigned to that operation will be hatched. Advantage: faster on the UI, as the hatches don't need to be recalculated on element changes or movements. They will only be applied during the burn phase (have a look at the simulation to see how they would turn out)

[image]

jpirnay, 2025-04-21:
The angle calculation was wrong inside the parameter test - this has been addressed.

closed 2025-05-29 by jpirnay