meerk40t/meerk40t #2855: [Bug]: Ruida control GRBL control problems

IngeNoodle, 2025-04-23:
### Summary Description

hello! i have a K40 laser with a lihuiyu m2 nano board and im using meerk40t as a bridge to my unit.

my problem started when the ruidacontrol began to make a loop when i send a dimple sqare or a circle. the head of the laser just moves in the image area and doesnt start the burning process. when i press the  stop button in lightburn the machine stops but when i try to resend the same job the machine doesnt move. to solve this i need to shut down and close all the programs and units and restart anyting.

with GRBLControl the ruida bugs appear to be fixed (so lightburn is not the problem) but when the work or the first layer ends the pointer just grab another start point and overburn the finished area.

after that, when the pointer go to home but it try to go offbounds on the X-Axis, like the software forgot where is the pointer.

as an extra. by some reason, when you press "get position" in lightburn, the GRBL doesnt deliver any coords.

what can i do to solve this?

### Crash logs

```shell

```

### MeerK40t Version

0.9.73

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2025-04-23:
Can you share the lb-file you are using?

IngeNoodle, 2025-04-23:
> Can you share the lb-file you are using?

[lbfile.zip](https://github.com/user-attachments/files/19876471/lbfile.zip)

there!. i'll be uploading some videos of the issue.
the version of lightburn is 1.7.08

IngeNoodle, 2025-04-24:
https://youtube.com/shorts/CP8R6wVYRAY

this is a video i recorded about the bug.
at this time, the program just stopped their bucle and throw a crash report. 

i cant copy any details of the crash report becaise the own window doesnt allow me to do that but i hoipe the error on the video where enought to give some guide to solve this

jpirnay, 2025-04-24:
This a serial number challenge issue, which is a bit strange. Are you able to properly burn from within meerk40t (without lightburn)? Whats you board version and what are the settings for the k40 within meerk40t?

IngeNoodle, 2025-04-24:
> This a serial number challenge issue, which is a bit strange. Are you able to properly burn from within meerk40t (without lightburn)? Whats you board version and what are the settings for the k40 within meerk40t?

yes. i'm able to burn from meerk40t without problems but the main reason i'm using LB its because their image processing.

i can use LB with GRBLControl and all works perfectly except for the problem of the misplaced layers.

this are my board details:
FW:  2019.01.12
model: 6c6879-laser-m2:9

im using the driver from corel laser hack.

i'll be uploading today the fault about the GRBL control

jpirnay, 2025-04-24:
I am not able to recreate the issues you are describing (lb 1.7.04):
lb via ruida https://streamable.com/afqonc
lb via grbl and two layers https://streamable.com/q32q1b
I wont be around for the next couple of days, so this will be dormant a bit I am afraid

IngeNoodle, 2025-04-24:
> I am not able to recreate the issues you are describing (lb 1.7.04): lb via ruida https://streamable.com/afqonc lb via grbl and two layers https://streamable.com/q32q1b I wont be around for the next couple of days, so this will be dormant a bit I am afraid

what kind or type of driver are you using?, what version of python? a special configuration to set?
i see you're using a differente version of LB, all this problem began with the latest update, so i think i've need to downgrade to your version.

i'll be trying it today and posting results.

IngeNoodle, 2025-04-24:
ok im just in front of the machine and now meerk40t throw this error just trying to send a single circle to burn

[MeerK40t-2025-04-24_16_44_54.txt](https://github.com/user-attachments/files/19900296/MeerK40t-2025-04-24_16_44_54.txt)

also, somehow the conecction drops away in the middle of the cut

this is the USB log
    Using LibUSB to connect.
    Attempting connection to USB.
    
    
    Finding devices.
    1 device(s) were found.
    K40 device detected:
    DEVICE ID 1a86:5512 on Bus 000 Address 001 =================
    	 bLength                :   0x12 (18 bytes)
    	 bDescriptorType        :    0x1 Device
    	 bcdUSB                 :  0x110 USB 1.1
    	 bDeviceClass           :   0xff Vendor-specific
    	 bDeviceSubClass        :    0x0
    	 bDeviceProtocol        :    0x2
    	 bMaxPacketSize0        :    0x8 (8 bytes)
    	 idVendor               : 0x1a86
    	 idProduct              : 0x5512
    	 bcdDevice              :  0x304 Device 3.04
    	 iManufacturer          :    0x0 
    	 iProduct               :    0x0 
    	 iSerialNumber          :    0x0 
    	 bNumConfigurations     :    0x1
    	  CONFIGURATION 1: 100 mA ==================================
    	   bLength              :    0x9 (9 bytes)
    	   bDescriptorType      :    0x2 Configuration
    	   wTotalLength         :   0x27 (39 bytes)
    	   bNumInterfaces       :    0x1
    	   bConfigurationValue  :    0x1
    	   iConfiguration       :    0x0 
    	   bmAttributes         :   0x80 Bus Powered
    	   bMaxPower            :   0x32 (100 mA)
    	    INTERFACE 0: Vendor Specific ===========================
    	     bLength            :    0x9 (9 bytes)
    	     bDescriptorType    :    0x4 Interface
    	     bInterfaceNumber   :    0x0
    	     bAlternateSetting  :    0x0
    	     bNumEndpoints      :    0x3
    	     bInterfaceClass    :   0xff Vendor Specific
    	     bInterfaceSubClass :    0x1
    	     bInterfaceProtocol :    0x2
    	     iInterface         :    0x0 
    	      ENDPOINT 0x82: Bulk IN ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x82 IN
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x2: Bulk OUT ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :    0x2 OUT
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x81: Interrupt IN ==========================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x81 IN
    	       bmAttributes     :    0x3 Interrupt
    	       wMaxPacketSize   :    0x8 (8 bytes)
    	       bInterval        :    0x1
    Config Set
    Config Set: Success
    Getting Active Config
    Active Config: Success.
    Kernel detach: Not Implemented.
    Attempting to claim interface.
    Interface claim: Success
    USB Connected.
    Sending CH341 mode change to EPP1.9.
    CH341 mode change to EPP1.9: Success.
    Device Connected.
    
    CH341 Chip Version: 50
    Driver Detected: LibUsb
    Device Connected.
    
    [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: Uno de los dispositivos conectados al sistema no funciona.\r\n\n'
    [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'

tiger12506, 2025-04-25:
What command-line are you using to run ruidacontrol and/or grblcontrol?
Can you try adding a -c, which will open a meerk40t console in the terminal. This would be a stab in the dark as a solution for connection dropping away in the middle of a cut.

IngeNoodle, 2025-04-26:
> What command-line are you using to run ruidacontrol and/or grblcontrol? Can you try adding a -c, which will open a meerk40t console in the terminal. This would be a stab in the dark as a solution for connection dropping away in the middle of a cut.

hi!. today i was checing my laptop and all the connection dropping was a fault from the port where i was connecting the  k40.

by today i was able to open LB files and send to cut with the layers configuration.
yesterday i was trying to run with ruida or grbl control but the initial bug (tue looped movement) persist.

i think it's a fault of my board or something else because at first try i was able to cut anything.