meerk40t/meerk40t #2865: [Enhancement] Undo/Redo list

Sophist-UK, 2025-05-20:
Currently looks like:
[image]

There are two issues IMO:

1. List is from earliest to latest but that isn't easily obvious

2. When you click on an item it undoes that item and all items later than that - which means that Undo and Redo operate differently and intuitively (even if logically consistent even if less intuitive). Most other software always returns to the state AFTER the action you click on.

I suggest that the layout and functionality of this is changed so that either:

### My preferred option
* Undo and Redo are in bold and not light grey
* Undo section starts with a greyed out option `Earliest` and the Redo section end with greyed out `Latest`
* Last undo item is greyed out and suffixed with `(current)`
* Clicking on the item returns to the state chronologically after the item clicked.

### Alternative option
* Retain the current functionality and only change visual layout
* Have 3 sections - as first option plus a single central section with a single greyed out item `Current`

jpirnay, 2025-05-23:
Thank you for your input. I am unclear though about the details.
The technical implementation stores the tree content *before* the labelled operation, so eg 'rectangle' stores the tree before adding the rectangle.
An undo of said entry 'undoes the rectangle', a redo of it restores the tree content after the operation, ie 'redoes the rectangle'.
So i think the actions+their labeling 'make sense' (to me).
The sequence or the lack of its denomination is a different story, though, which could either be made clearer by enumerating them or taking your idea of adding a 'current' entry on board.
Any thoughts?

Sophist-UK, 2025-05-23:
I was comparing it to how Inkscape does its undo lost. When you click on the entry you end up with the state AFTER that entry. So the bottom most entry on the Undo list (not greyed out) is the current status and after than the Redo items are greyed out or italicised.

So whilst it is an undo/redo list in function, it is presented as a set of states AFTER an action that you can select from.

jpirnay, 2025-05-23:
If you look at other software (ie office) then the behaviour is different, clicking on the undo history brings you to the state *before* the labeled action. Seems to be a question of terminology.

closed 2025-06-16 by jpirnay