meerk40t/meerk40t #2888: [Bug]: Laser is on while moving on the Y-Axis

webfraggle, 2025-06-01:
### Summary Description

Hello,

I have a strange problem. Often, but not anytime the laser is firing while moving from one vector to another on the Y-Axis. 
You can see it in this picture of my last cut.
I have a monport 45 with an lihuiyu m3 board. The purple one.

[image]

Any ideas how I can avoid this?

Thank you very much.

Christoph

### Crash logs

```shell

```

### MeerK40t Version

0.9.7030

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2025-06-01:
That's an *interesting* one... Can you be more specific about the M3-nano version - what does the board say exactly ? I am looking for the fireware version imprinted on the board.

webfraggle, 2025-06-01:
Thanks for your fast reply.

Here are some more Details.

The long cut was made before starting the cut, only by using the function "Align laser with the center of the selection". 

Fireware Ver: 2022.03.18
Model: 6C6879-LASER-M3:10

[image]

[image]
[image]

webfraggle, 2025-06-18:
Hello @jpirnay 

I think it's great that you have ordered an m3nano. 
I have cut something several times today and the above error has not occurred.
However, I also tried using K40 Whisperer. I noticed that the pure movement of the axes is much smoother than with Meerk40t. You can see and hear this. That's why I uploaded a video about it.

[https://youtu.be/GCE5jfpBfHg](https://youtu.be/GCE5jfpBfHg)

I also noticed that if you move back and forth too fast, it sometimes stalls or somehow goes over the point and then the cuts are no longer precise. I'm currently moving at a maximum of 20 mm/s.

I would be incredibly happy if you could find a solution.

I also tried the mLaserArt supplied with the machine, which also made the process smoother.

Unfortunately, I couldn't make a cut with either, as both had different problems with Inkscape.

And one more question. The m3nano has power control via PWM. 
https://www.scorchworks.com/Blog/laser-power-control-in-k40-whisperer-with-a-m3-nano-controller-board/

Could this be built into Meerk40t? Then maybe I would also convert my laser.

Thanks again for your efforts

Translated with DeepL.com (free version)

Sophist-UK, 2025-06-19:
I watched the video and the sound difference is indeed noticeable. Are the movement speeds the same?

webfraggle, 2025-06-21:
This is the sequence: Home, down, left, right, home. 

Settings: 
Board: M3
Jog distance 10mm
X Travel Speed: 20mm/s
Y Travel Speed: 15mm/s

`
[16:41:13]     Window opened: Controller
b'\x00IPPFFFFFFFFFFFFFFFFFFFFFFFFFFF\xe4'
b'\x00ICV2490731016000027CNBRS1Ez139\x9d'
b'\x00FNSEFFFFFFFFFFFFFFFFFFFFFFFFFFB'
b'\x00ICV2502451021000015CNRBS1Ez1394'
b'\x00FNSEFFFFFFFFFFFFFFFFFFFFFFFFFFB'
b'\x00ICV2502451021000015CNRTS1Ez139\xf3'
b'\x00FNSEFFFFFFFFFFFFFFFFFFFFFFFFFFB'
b'\x00IPPFFFFFFFFFFFFFFFFFFFFFFFFFFF\xe4'
`

webfraggle, 2025-06-29:
But that's a strange phenomenon. However, there is still the original problem that the laser is randomly switched on when y-moving. I tried a bit more and couldn't reproduce it reliably. It seems random.
Could it be due to the initialise function? Could it be that commands are being sent to the controller incorrectly? Perhaps due to an incorrect connection speed?
I don't have this problem in K40whisperer.
Is there a way how I can debug the USB communication in both tools (Meerk40t and K40Whisperer) to detect differences?
Thank you very much for your efforts

[image]

Translated with DeepL.com (free version)

jpirnay, 2025-06-29:
And just for the sake of completeness: you are showing vector engraves/cuts not rasters?

webfraggle, 2025-06-29:
I have now taken the last version from today from the main branch. And I haven't had any errors in six tests so far. I have removed the speed limit and limit write buffer. 
And yes, I am using Vector operations, and yes I have set the M3 board. See also the screenshot above.
Attached is my .cfg file.

```
[/]
single_instance_only = True
high_dpi = True
forced_theme = light
tooltip_autopop = 10000
tooltip_delay = 100
align_first = True
developer_mode = False
debug_mode = False
beep_soundfile = c:\Windows\Media\Alarm01.wav
draw_mode = 0
inkscape_path = C:/Program Files/Inkscape/bin/inkscape.exe
svg_not_supported = 0
system_font_directories = ['C:\\Windows\\Fonts', 'C:\\Users\\chris\\AppData\\Local\\Microsoft\\Windows\\Fonts']
update_check = 1
update_frequency = 1
check_for_betas = False
print_shutdown = False
preferred_device = lhystudios
units_name = mm
supress_non_visible = True
windows_save = True
auto_spooler = True
mouse_wheel_pan = False
mouse_pan_invert = False
mouse_zoom_invert = False
disable_tool_tips = False
disable_tree_tool_tips = False
concern_level = 0
user_scale = 100
faulty_bitmap_scaling = False
bitmap_correction_scale = 1.0
enable_sel_move = True
enable_sel_size = True
enable_sel_rotate = True
enable_sel_skew = False
zoom_margin = 4
ribbon_appearance = default
icon_size = small
mini_icon = False
tree_colored = True
zoom_factor = 0.1
pan_factor = 25.0
autofocus_resize = False
suppress_focus_animation = False
select_smallest = True
auto_select = True
outer_handles = False
coord_display = 0
show_attract_len = 45
snap_points = False
action_attract_len = 20
snap_grid = True
grid_attract_len = 15
clear_magnets = True
button_repeat = 0.5
button_accelerate = True
just_a_single_element = False
process_while_typing = False
process_while_sliding = False
pane_lock = False
console_font = 
history_limit = 50
confined = True
jog_amount = 10.0mm
[... 619 lines cut ...]
+rotate_ccw = .timerrotate_ccw 0 0.1 rotate -2
+translate_right = .timertranslate_right 0 0.1 translate 1mm 0
+translate_left = .timertranslate_left 0 0.1 translate -1mm 0
+translate_down = .timertranslate_down 0 0.1 translate 0 1mm
+translate_up = .timertranslate_up 0 0.1 translate 0 -1mm
+right = .timerright 0 0.1 right 1mm
+left = .timerleft 0 0.1 left 1mm
+up = .timerup 0 0.1 up 1mm
+down = .timerdown 0 0.1 down 1mm
burn = planz clear copy preprocess validate blob preopt optimize spool
-scale_up = .timerscale_up off
-scale_down = .timerscale_down off
-rotate_cw = .timerrotate_cw off
-rotate_ccw = .timerrotate_ccw off
-translate_right = .timertranslate_right off
-translate_left = .timertranslate_left off
-translate_down = .timertranslate_down off
-translate_up = .timertranslate_up off
-right = .timerright off
-left = .timerleft off
-up = .timerup off
-down = .timerdown off
reset_bind_alias = bind default;alias default

```

webfraggle, 2025-07-05:
At the moment, I cannot reproduce this bug with the source version from github, so I close this bug. 
Hope that's ok.

closed 2025-07-05 by webfraggle