meerk40t/meerk40t #2931: [Bug]: m3 nano pwm and ruidacontrol

slayeralive, 2025-07-16:
### Summary Description

I connected m3 nano purple board to power supply for pwm control and it works great in meerk40t. Problem is i cant use Lightburn anymore and i have a lot if projects saved as .lbm2.
Using Ruidacontrol doesn't fire the laser at all only moves the machine, grblcontrol fires the laser only if power set in lb is at 100%, any lower and it doesn't fire at all.

### Crash logs

```shell

```

### MeerK40t Version

0.98b2

### MeerK40t Type

Executable

### Your Operating System

Windows

incredibledeath, 2025-07-17:
> Fixed in PR [#2932](https://github.com/meerk40t/meerk40t/pull/2932)

@jpirnay Great! Thanks a lot! How can i run the new version? Im on windows, is there an executable? Sorry im not a programmer i dont know how to run different versions from source, i can follow a guide.

jpirnay, 2025-07-17:
There's a new beta release available