meerk40t/meerk40t #2954: [Suggestion]: GCode file export

Zumpanoa, 2025-07-29:
### Summary Description

Have you foreseen the possibility of exporting the work to a GCode file to be sent to the laser at a later time using a device such as the SCULPFUN TS1 touch screen?

### MeerK40t Version

v0.9.7940

### MeerK40t Type

Executable

### Your Operating System

Windows

SilvanoCC, 2025-08-08:
Io uso  touch screen SCULPFUN TS1  ma con il codice iso così come viene generato non mi funziona . 
Devo modificarlo manualmente e aggiungere delle istruzioni alla fine e all'inizio.
G00 G17 G40 G21 G54
G90
M4
M8

Parte di iso lavoro

G0 X0Y0
M2

Ci sarebbe il modo di poter modificare per così dire il postprocessor per far si che vengano inserite in automatico?

jpirnay, 2025-08-08:
Is it to make it run at all or for the export?
Within meerk40t you have:
- GRBL startup commands: 
[image]

- Default operations to add at job start / job end, eg you cold automatically add a HOME command at the end of the job or a console command at the start of the job (`` ``) - this will be exported too.
[image]

SilvanoCC, 2025-08-09:
Ho provato ad aggiungere il " gcode" nella console , ma quando salvo si meerk40t si blocca e non salva.
Se invece provo a farlo senza la console , ma anche con funzione Home va.

jpirnay, 2026-01-02:
The bug was a bit confusing, as it turned out the export job output was following another path than the output of coolant changes. 
I am suspecting you are using MeerK40t just as gcode-creator? Your interface for the device is probably not properly configured, so that the coolant commands would try to interact with a non-working device configuration and hang. If that's the case then you should define the device interface as mock, to avoid unwanted side-effects.
In any case should be working now regardless (PR #3145 )

closed 2026-01-07 by jpirnay