meerk40t/meerk40t #3019: [Bug]: commit a7e04ae6b8abadb57d77d6719cf6828270d06e77 is the last one working (for me)

kostex, 2025-09-15:
### Summary Description

Hi, After a week of not using Meerk40T (Atomstack A10V2 user)  and using the git repo version, Meerk40t would do homing/moving/connecting just fine but a burn would not start when opening existing or importing svg files.
Testing with default polygons (line/circle/rectangle) from Meerk40t, burns DO work, but imported shapes (svg) won't. The background task popup stops after:
planz8 clear copy preprocess validate blob preopt optimize spool (preprocessing) stage.

After checking out different stages I've found that
commit a7e04ae6b8abadb57d77d6719cf6828270d06e77 (Add shapely)
is the last commit that had no issues.
bee63ea3e261eaf96048d826cf213148578c9da7 (Fix/speedup of is_inside function) is the culprit.

Hope you have enough information to investigate.
Thanks,
Roel

### Crash logs

```shell

```

### MeerK40t Version

GIT Latest commits

### MeerK40t Type

Git

### Your Operating System

Linux

jpirnay, 2025-09-15:
Thank you for reporting it, should be fixed, ie reverted, please check

kostex, 2025-09-16:
Thank you very much!
Problem solved!