meerk40t/meerk40t #3027: Support of AlgoLaser products

RiccardoC91, 2025-09-25:
Hi,
I've bought a AlgoLaser MK2 laser, that works well with LaserGRBL but not with Meerk40t.
I cannot configure the device correctly, when I use the button Query Properties in the configuration panel there is no response, and the problem occurs with both serial or networked connection.
Sometimes the application also crashes, this is an example of error (I've the version 0.9.8100 and I use windows 11)

MeerK40t crash log. Version: 0.9.6400 on Windows: Python 3.8.10: AMD64 - wxPython: 4.2.2 msw (phoenix) wxWidgets 3.2.6
Traceback (most recent call last):
  File "meerk40t\kernel\kernel.py", line 1779, in run
  File "meerk40t\grbl\controller.py", line 663, in _recving
  File "meerk40t\grbl\tcp_connection.py", line 68, in read
ConnectionResetError: [WinError 10054] Connessione in corso interrotta forzatamente dall'host remoto

Local variables:
[meerk40t\kernel\kernel.py:1779, in run]:
func_result : None
sys : <module 'sys' (built-in)>
_ : <built-in function GetTranslation>
args : ()
channel : Channel('threads', buffer_size=0, line_end=None)
func : <bound method GrblController._recving of GRBLController('192.168.5.219:23')>
result : <bound method GrblController._rstop of GRBLController('192.168.5.219:23')>
self : Kernel(MeerK40t, MeerK40t, 0.9.6400)
thread_name : recver-192.168.5.219:23
[meerk40t\grbl\controller.py:663, in _recving]:
self : GRBLController('192.168.5.219:23')
response : None
cmd_issued : $

[meerk40t\grbl\tcp_connection.py:68, in read]:
self : TCPOutput('192.168.5.219:23')
f : -1

There is the possibility to support also this kind of laser?

Thank you for all the response

jpirnay, 2025-09-29:
I am working on a more general initialization routine, but you could try to circumvent one possible blocker by disabling the Require Validator setting in the device settings (tab Validation)

RiccardoC91, 2025-09-29:
Hi jpirnay,
the problem is the same also with the USB connection.
In both cases, if I open the controller tab and i press Reset, I receive the response message from the laser

[image]

and if I press Home, the laser goes to the home position.
But this is all, no other commands works.

I've already disabled the Validator, but nothing changes.

[image]

Thank you for the support!

jpirnay, 2025-09-29:
This is a grblhal controller, not something I have encountered so far, looks like as if meerk40t would wait for a OK response that isnt happening. I need to see whether I can flash a esp32 controller with a grblhal version.
What would be interesting to see what the controller window displays if you click for instance on the jog right button

RiccardoC91, 2025-09-30:
If I try to move the laser, nothing is written in the controller window. If I press the lock or unlock buttos, the application crashes. This is the error for the lock operation:

MeerK40t crash log. Version: 0.9.8100 on Windows: Python 3.8.10: AMD64 - wxPython: 4.2.2 msw (phoenix) wxWidgets 3.2.6
Traceback (most recent call last):
  File "meerk40t\kernel\kernel.py", line 1801, in run
  File "meerk40t\core\spoolers.py", line 544, in run
  File "meerk40t\core\laserjob.py", line 103, in execute
  File "meerk40t\core\laserjob.py", line 172, in execute_item
AttributeError: 'GRBLDriver' object has no attribute 'lock_rail'

Local variables:
[meerk40t\kernel\kernel.py:1801, in run]:
func_result : None
sys : <module 'sys' (built-in)>
_ : <built-in function GetTranslation>
args : ()
channel : Channel('threads', buffer_size=0, line_end=None)
func : <bound method Spooler.run of Spooler(AlgoLaser)>
result : <function Spooler.restart.<locals>.clear_thread at 0x25BEABF8>
self : Kernel(MeerK40t, MeerK40t, 0.9.8100)
thread_name : Spooler(grbl)
[meerk40t\core\spoolers.py:544, in run]:
self : Spooler(AlgoLaser)
is_valid : True
idx : 0
program : LaserJob(('lock_rail',): 0/1)
priority : 0
fully_executed : True
[meerk40t\core\laserjob.py:103, in execute]:
self : LaserJob(('lock_rail',): 0/1)
driver : GRBLDriver(AlgoLaser)
item : ('lock_rail',)
[meerk40t\core\laserjob.py:172, in execute_item]:
item : ('lock_rail',)
_handle_console_direct : <function LaserJob.execute_item.<locals>._handle_console_direct at 0x2CD772B0>
self : LaserJob(('lock_rail',): 0/1)

jpirnay, 2025-10-02:
I initially thought it would be grblhal firmwares in general, but I did setup am mks dlc32 v2 board with grblhal and it does communicate properly with meerk40t.
```
-->	[AXS:3:XYZ]
-->	[NEWOPT:ENUMS,RT+,PROBES=1,ES,REBOOT,TC,SED,RTC,WIFI,FS,SD]
-->	[FIRMWARE:grblHAL]
-->	[SIGNALS:EP]
-->	[NVS STORAGE:*FLASH 4K]
-->	[FREE MEMORY:161K]
-->	[DRIVER:ESP32]
-->	[DRIVER VERSION:250927]
-->	[DRIVER OPTIONS:4.4.2]
-->	[BOARD:MKS DLC32 2.x]
-->	[AUX IO:2,2,0,0]
-->	[WIFI AP MAC:78:ee:4c:e5:97:b5]
-->	[AP IP:192.168.5.1]
-->	[PLUGIN:FS stream v1.01]
-->	[PLUGIN:FS macro plugin v0.19]
-->	[PLUGIN:WebUI v0.27]
-->	[PLUGIN:SDCARD v1.24]
```

So what type of board / what version of grblhal is this?